ros-simulation/gazebo_ros_pkgs #1169: test_node is flaky

jacobperron, 2020-09-23:
I first noticed the test failing here: https://build.osrfoundation.org/job/ros2_gazebo_pkgs-ci-default_rolling-focal-amd64/5/

Locally, I can reproduce if I run the test standalone (not with `colcon test`). I think the lack of stress is uncovering some sort of race in the test:

```
$ ./build/gazebo_ros/test/test_node
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from TestNode
[ RUN      ] TestNode.StaticNode
[INFO] [1600895033.746531687] [gazebo_ros_node]: ROS was initialized without arguments.
[       OK ] TestNode.StaticNode (58 ms)
[ RUN      ] TestNode.GetSdf
Warning [parser.cc:636] Converting a deprecated SDF source[data-string].
[INFO] [1600895033.852026918] [gazebo_ros_node]: ROS was initialized without arguments.
[       OK ] TestNode.GetSdf (122 ms)
[ RUN      ] TestNode.RemapAndQoSOverride
Warning [parser.cc:636] Converting a deprecated SDF source[data-string].
[INFO] [1600895033.976637521] [gazebo_ros_node]: ROS was initialized without arguments.
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create subscription: rcl node's context is invalid, at /tmp/binarydeb/ros-rolling-rcl-2.2.0/src/rcl/node.c:426
[1]    331222 abort (core dumped)
```

<details>
<summary>Backtrace</summary>
<pre>
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f6992d7d859 in __GI_abort () at abort.c:79
#2  0x00007f6993152951 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f699315e47c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f699315e4e7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f699315e46f in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f69935af6ed in rclcpp::exceptions::throw_from_rcl_error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_error_state_t const*, void (*)()) () at /opt/ros/rolling/lib/librclcpp.so
#7  0x00007f699363c7a6 in rclcpp::SubscriptionBase::SubscriptionBase(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t const&, bool) () at /opt/ros/rolling/lib/librclcpp.so
#8  0x00007f699365f8ed in rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >::Subscription(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<rosgraph_msgs::msg::Clock_<std::allocator<void> > > >) () at /opt/ros/rolling/lib/librclcpp.so
#9  0x00007f6993660764 in rclcpp::create_subscription_factory<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::_Bind<void (rclcpp::TimeSource::*(rclcpp::TimeSource*, std::_Placeholder<1>))(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > >)>, std::allocator<void>, rosgraph_msgs::msg::Clock_<std::allocator<void> >, rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >(std::_Bind<void (rclcpp::TimeSource::*(rclcpp::TimeSource*, std::_Placeholder<1>))(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > >)>&&, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >::SharedPtr, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<rosgraph_msgs::msg::Clock_<std::allocator<void> > > >)::{lambda(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&)#1}::operator()(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&) const () at /opt/ros/rolling/lib/librclcpp.so
#10 0x00007f6993660948 in std::_Function_handler<std::shared_ptr<rclcpp::SubscriptionBase> (rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&), rclcpp::create_subscription_factory<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::_Bind<void (rclcpp::TimeSource::*(rclcpp::TimeSource*, std::_Placeholder<1>))(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > >)>, std::allocator<void>, rosgraph_msgs::msg::Clock_<std::allocator<void> >, rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >(std::_Bind<void (rclcpp::TimeSource::*(rclcpp::TimeSource*, std::_Placeholder<1>))(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > >)>&&, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >::SharedPtr, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<rosgraph_msgs::msg::Clock_<std::allocator<void> > > >)::{lambda(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&)#1}>::_M_invoke(std::_Any_data const&, rclcpp::node_interfaces::NodeBaseInterface*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&) () at /opt/ros/rolling/lib/librclcpp.so
#11 0x00007f699360182b in rclcpp::node_interfaces::NodeTopics::create_subscription(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::SubscriptionFactory const&, rclcpp::QoS const&) () at /opt/ros/rolling/lib/librclcpp.so
#12 0x00007f699365d479 in std::shared_ptr<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > > rclcpp::create_subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::_Bind<void (rclcpp::TimeSource::*(rclcpp::TimeSource*, std::_Placeholder<1>))(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > >)>, std::allocator<void>, rosgraph_msgs::msg::Clock_<std::allocator<void> >, rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>&>(std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, std::_Bind<void (rclcpp::TimeSource::*(rclcpp::TimeSource*, std::_Placeholder<1>))(std::shared_ptr<rosgraph_msgs::msg::Clock_<std::allocator<void> > >)>&&, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >::SharedPtr) () at /opt/ros/rolling/lib/librclcpp.so
[... cut ...]
#32 0x00007f699375ce00 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::_Bind<void (gazebo_ros::Executor::*(gazebo_ros::Executor*))()> > > >::_M_run() () at /home/jacob/ws/ros/gazebo_ros_pkgs/install/gazebo_ros/lib/libgazebo_ros_node.so
#33 0x00007f699318acb4 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#34 0x00007f699347f609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#35 0x00007f6992e7a293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
</pre>
</details>

jacobperron, 2020-09-23:
I'm looking into a fix.

jacobperron, 2020-09-24:
Possibly related: I don't think `rclcpp::MultiThreadedExecutor` is thread-safe, which is possible causing this (or similar errors).

jacobperron, 2021-01-06:
Using ros2/rtclcpp#1505, which addresses thread-safety issues, I still get the same crash in `test_node`. Here's a backtrace:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f13342fa859 in __GI_abort () at abort.c:79
#2  0x00007f13346cf951 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f13346db47c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f13346db4e7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f13346db46f in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f1334ffe57c in rclcpp::exceptions::throw_from_rcl_error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_error_state_t const*, void (*)())
    (ret=200, prefix="could not create subscription", error_state=0x0, reset_error=0x7f133481c28d <rcutils_reset_error>)
    at /home/jacob/ws/ros/latest/src/ros2/rclcpp/rclcpp/src/rclcpp/exceptions/exceptions.cpp:90
#7  0x00007f1335125fd9 in rclcpp::SubscriptionBase::SubscriptionBase(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t const&, bool) (this=0x7f1310001340, node_base=
    0x560838f5f9f0, type_support_handle=..., topic_name="/clock", subscription_options=..., is_serialized=false)
    at /home/jacob/ws/ros/latest/src/ros2/rclcpp/rclcpp/src/rclcpp/subscription_base.cpp:80
#8  0x00007f133514d059 in rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >::Subscription(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<rosgraph_msgs::msg::Clock_<std::allocator<void> > > >)
    (this=0x7f1310001340, node_base=0x560838f5f9f0, type_support_handle=..., topic_name="/clock", qos=..., callback=..., options=..., message_memory_strategy=std::shared_ptr<class rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >> (use count 4, weak count 0) = {...}, subscription_topic_statistics=std::shared_ptr<class rclcpp::topic_statistics::SubscriptionTopicStatistics<rosgraph_msgs::msg::Clock_<std::allocator<void> > >> (empty) = {...})
    at /home/jacob/ws/ros/latest/src/ros2/rclcpp/rclcpp/include/rclcpp/subscription.hpp:119
#9  0x00007f133514b63e in __gnu_cxx::new_allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > >::construct<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&, std::shared_ptr<rclcpp::topic_statistics::Subscr--Type <RET> for more, q to quit, c to continue without paging--
ck_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >*, rclcpp::node_interfaces::NodeBaseInterface*&, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<rosgraph_msgs::msg::Clock_<std::allocator<void> > > > const&) (this=0x7f1301ff61ff, __p=0x7f1310001340) at /usr/include/c++/9/ext/new_allocator.h:147
#10 0x00007f1335149b7a in std::allocator_traits<std::allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > > >::construct<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void>, rclcpp [... cut ...]
#49 0x00007f133501d708 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::_Bind<void (rclcpp::executors::MultiThreadedExecutor::*(rclcpp::executors::MultiThreadedExecutor*, unsigned long))(unsigned long)> > > >::_M_run() (this=0x7f12fc001f10)
    at /usr/include/c++/9/thread:195
#50 0x00007f1334707d84 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#51 0x00007f1334837609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#52 0x00007f13343f7293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

ivanpauno, 2021-01-06:
I will try to reproduce, as it doesn't seem obvious what's the problem from the traceback.
It seems that a node is being destroyed while the executor is still running and a subscription is still around, but I'm not sure how that can happen.

PS:

```
    (ret=200, prefix="could not create subscription", error_state=0x0, reset_error=0x7f133481c28d <rcutils_reset_error>)
```

`ret=200` is an [invalid node](https://github.com/ros2/rcl/blob/38069b72e75b747164efd3e4eecd499bf008279b/rcl/include/rcl/types.h#L52).

jacobperron, 2021-02-03:
Fixed by https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1212

closed 2021-02-03 by jacobperron