ros-simulation/gazebo_ros_pkgs #1174: gazebo_ros_depth_camera not publishing depth camera_info

corot, 2020-10-02:
The code is commented both in melodic and noetic: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_plugins/src/gazebo_ros_depth_camera.cpp#L477

This simple PR fixes the issue: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/97

HighPriest, 2020-10-12:
There seem to be a clusterfuck of issues and merges related to camera plugin.
It's baffling how such a prelevant sensor is so badly documented and maintained. eg. #127 is still hanging even though the issue has been resolved

Anyway, I am having the same issue on ROS2 Foxy, Gazebo11.12, no data is published by (in this case) gazebo_ros_camera.

chapulina, 2020-10-13:
Contributions are always welcome. This includes not only ticketing issues, which is very appreciated, but also opening and reviewing pull requests, commenting on issues that are resolved...

@corot , #97 was merged well before the Kinetic / Melodic / Noetic releases. If you can figure out what the fix is for the newer distros, a PR would be appreciated.

@HighPriest , I believe you're a bit confused. This issue is about the depth camera on ROS 1, and it looks like you're having some troubles with the camera on ROS 2? Please open a new issue explaining your exact issue and how to reproduce it. Or even better, open a pull request proposing a fix. The camera info was working when first ported to ROS 2 on https://github.com/ros-simulation/gazebo_ros_pkgs/pull/827.

HighPriest, 2020-10-13:
@chapulina ros_depth_camera has been merged with other vision related plugins into single ros_camera. The goal behind it was to differentiate them with sensor type.

chapulina, 2020-10-13:
> ros_depth_camera has been merged with other vision related plugins into single ros_camera

Oh yeah you're right. In any case, I think the ROS 2 issue, if there's any, should be independent of this one. I still don't know how to reproduce your issue.

---

I completely understand your frustration, there's a lot to be improved. But the way you've approached this issue isn't productive. I recommend you take a look at our [code of conduct](https://osrf.github.io/code_of_conduct/).

HighPriest, 2020-10-15:
@chapulina We have found a workaround for the broken camera and camera_depth in gazebo11 / ros_foxy,
We took source of gazebo_ros_pkgs/dashing and built the whole dir gazebo_plugins against gazebo11 and ros_foxy (both from package manager). Probably it's enough to just build gazebo_ros_camera and gazebo_ros_depth_camera from gazebo_plugins

Then we attached the newly built .so files to the robot definition. (filename="absolute/path/to/gazebo_ros_depth_camera.so") [in our case we want depth] and all vision data is now correctly pushed to ros topic.

chapulina, 2020-10-15:
Thanks for the update, @HighPriest  :+1:  It would be interesting to know exactly what's the difference between the `dashing` and `ros2` branches that's making that work.

HighPriest, 2020-10-15:
@chapulina 
There were so many changes in between dashing and foxy release (including some damn PROFILER that makes the code a mashup of crap, that none of my teammates are able to decipher), that make us unable to understand the code and debug the issue. But I don't think anyone is able to rollback from the PROFILER implementation.

From our initial debugging session, we have found that the "publish" function in current (gazebo_ros_pkgs/ros2) gazebo_ros_camera is reached and executed, no data ends up on the topic.

corot, 2020-12-02:
Not 100% sure, but we may be mixing 2 different issues here. One is that gazebo_ros_depth_camera not publishing depth camera_info on melodic, what gets fixed by re-making #97 with melodic.

The other is that, with the recent addition of the profiler to gazebo (on version 9.15), all camera plugins have stopped to work. To contour that 2nd problem, I'm now compiling gazebo 9.14 from sources,  but the camera info is still missing

corot, 2021-03-23:
Probably much clearer with a PR: #1260

corot, 2021-05-30:
I think this is not the same as #1175; that came from adding profiler to gazebo, afaik.  Here we are missing a publish, which was added to Kinetic but not melodic (the matching PR is #1260)