ros-simulation/gazebo_ros_pkgs #1176: Does this repository have too many branches?

joe28965, 2020-10-09:
There are 56 branches. 50 of them are stale. Granted, a few of these are branches like hydro/groovy/indigo branches, but it just feels a bit chaotic right now to be perfectly honest.

HighPriest, 2020-10-12:
It is. But both [osrf/gazebo](https://github.com/osrf/gazebo) and [ros2/ros2](https://github.com/ros2/ros2) are in the same state.

You could say it's normal for an open source project in rapid development, but like you have noticed, it's mostly stale, abandoned code.

chapulina, 2020-10-13:
I closed all the branches that had been merged. For a long time GitHub didn't have the auto-delete-branch-on-pull-request-merge feature, so some branches were just left there.

In any case, I believe the only branches that users should be concerned about are the release branches for each distro.

HighPriest, 2020-10-13:
Okay, but that's obvious for you, as a person accustomed with the Repo. Is there no way to pluck out the closed repositories?

chapulina, 2020-10-13:
I don't want to touch other people's branches. I've closed my branches that I thought could be closed. At this point, I think it would be difficult to foresee all the consequences of closing some branches, so it may be safer to keep them there in case someone in some corner of the world has been relying on them.

Maybe the confusion could be helped with documentation on the README about what the release branches are. i.e. `<distro>-devel` ones for ROS 1, `<distro>` for ROS 2, `ros2` for Foxy / Rolling.

Another place to check with precision what branch matches each distribution is https://github.com/ros/rosdistro, but that can be tedious.

closed 2022-04-06 by joe28965