ros-simulation/gazebo_ros_pkgs #1181: [ROS2] Can't load libgazebo_ros_imu_sensor

iggyrrieta, 2020-10-19:
I am using ROS2 Eloquent. I am able to load other plugins such as "libgazebo_ros_diff_drive" or "libgazebo_ros_joint_state_publisher" that are located at same folder.

I am using the plugin as found in the [wiki](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-IMU-Sensors) 

```xml
<link name="imu_sim_link">
    <sensor name="sensor_imu" type="imu">
        <always_on>true</always_on>
        <update_rate>30</update_rate>
        <plugin name="my_plugin_name" filename="libgazebo_ros_imu_sensor.so">
            <ros>
                <!-- Will publish to /imu/data -->
                <namespace>/imu</namespace>
                <argument>--ros-args --remap ~/out:=data</argument>
            </ros>
            <!-- frame_name ommited, will be "my_link" -->
        </plugin>
    </sensor>
</link>
```

Error Output:

`[Err] [Model.cc:1114] Model[myModel] is attempting to load a plugin, but detected an incorrect plugin type. Plugin filename[libgazebo_ros_imu_sensor.so] name[my_plugin_name]
`

ahcorde, 2020-10-19:
Hello @iggyrrieta,

can you try to run this world 
```
<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <!-- A global light source -->
    <include>
      <uri>model://sun</uri>
    </include>
    <!-- A ground plane -->
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <model name="box">>
      <link name="imu_sim_link">
        <collision name="collision">
          <geometry>
            <box>
              <size>1 1 1</size>
            </box>
          </geometry>
        </collision>
        <visual name="visual">
          <geometry>
            <box>
              <size>1 1 1</size>
            </box>
          </geometry>
          <material>
            <script>Gazebo/WoodPallet</script>
          </material>
        </visual>
        <sensor name="sensor_imu" type="imu">
            <always_on>true</always_on>
            <update_rate>30</update_rate>
            <plugin name="my_plugin_name" filename="libgazebo_ros_imu_sensor.so">
                <ros>
                    <!-- Will publish to /imu/data -->
                    <namespace>/imu</namespace>
                    <argument>--ros-args --remap ~/out:=data</argument>
                </ros>
                <!-- frame_name ommited, will be "my_link" -->
            </plugin>
        </sensor>
      </link>
    </model>
  </world>
</sdf>
```

Then you should run:

```bash
ros2 launch gazebo_ros gazebo.launch.py world:=<your world file>
```

Anyhow for future questions like this please use https://answers.gazebosim.org/questions/ and review the [FAQ](https://answers.gazebosim.org/faq/).

closed 2020-10-19 by ahcorde

iggyrrieta, 2020-10-19:
@ahcorde Thanks! Working! The problem was because of the way I create the `sdf` and `world` files. I first create a `xacro` file and then automatically convert it to `urdf` and `sdf`. Fixed joints or links without inertia are not created in the `sdf` file. But anyway modifying the `sdf` model and `world` the way you say works. I just thought that the conversion from `xacro` did all the work for me...

I am just writing this here, in case anyone got the same problem as me. Next I'll write in gazebosim.org/question, understood.