ros-simulation/gazebo_ros_pkgs #1182: A separate branch for Foxy and Rolling

joe28965, 2020-10-23:
Since June ROS 2 now has [Rolling Ridley](https://index.ros.org/doc/ros2/Releases/Release-Rolling-Ridley/). Currently the gazebo_ros ros2 branch is (if I remember correctly) linked to Foxy. Should there become a dedicated foxy branch, as there is a(n) dashing/eloquent branch, and have the ros2 branch as the rolling branch?

chapulina, 2020-10-26:
We shouldn't split branches unless we need to. Right now the `ros2` branch is working for both Foxy and Rolling:

* http://build.ros2.org/job/Fpr__gazebo_ros_pkgs__ubuntu_focal_amd64/
* http://build.ros2.org/job/Rpr__gazebo_ros_pkgs__ubuntu_focal_amd64/

Is there a breaking change coming up on Rolling that will affect `gazebo_ros_pkgs`?

joe28965, 2020-10-27:
Well, I am getting build errors when building `gazebo_ros`. Although they seem to come from a test in `gazebo_ros/test/test_plugins.cpp`. I am not sure if I'm the only one who has those errors, or if something is broken?

This is my gazebo_ros stderr.log from my build

`In file included from [01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67[m[K,
                 from [01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15[m[K:
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:38:[m[K [01;35m[Kwarning: [m[K‘[01m[Kconstexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()[m[K’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [[01;35m[K-Wdeprecated-declarations[m[K]
  496 |   static_assert(::testing::internal::[01;35m[KInstantiateTestCase_P_IsDeprecated[m[K(), \
      |                                      [01;35m[K^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[m[K
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:38:[m[K [01;36m[Knote: [m[Kin definition of macro ‘[01m[KINSTANTIATE_TEST_CASE_P[m[K’
  496 |   static_assert(::testing::internal::[01;36m[KInstantiateTestCase_P_IsDeprecated[m[K(), \
      |                                      [01;36m[K^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[m[K
In file included from [01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:62[m[K,
                 from [01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15[m[K:
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16:[m[K [01;36m[Knote: [m[Kdeclared here
 1210 | constexpr bool [01;36m[KInstantiateTestCase_P_IsDeprecated[m[K() { return true; }
      |                [01;36m[K^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[m[K
In file included from [01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67[m[K,
                 from [01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15[m[K:
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:73:[m[K [01;35m[Kwarning: [m[K‘[01m[Kconstexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()[m[K’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [[01;35m[K-Wdeprecated-declarations[m[K]
  496 |   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated([01;35m[K)[m[K, \
      |                                                                         [01;35m[K^[m[K
[01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1:[m[K [01;36m[Knote: [m[Kin expansion of macro ‘[01m[KINSTANTIATE_TEST_CASE_P[m[K’
   70 | [01;36m[KINSTANTIATE_TEST_CASE_P[m[K(
      | [01;36m[K^~~~~~~~~~~~~~~~~~~~~~~[m[K
In file included from [01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:62[m[K,
                 from [01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15[m[K:
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16:[m[K [01;36m[Knote: [m[Kdeclared here
 1210 | constexpr bool [01;36m[KInstantiateTestCase_P_IsDeprecated[m[K() { return true; }
      |                [01;36m[K^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[m[K
In file included from [01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67[m[K,
                 from [01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15[m[K:
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:73:[m[K [01;35m[Kwarning: [m[K‘[01m[Kconstexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()[m[K’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [[01;35m[K-Wdeprecated-declarations[m[K]
  496 |   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated([01;35m[K)[m[K, \
      |                                                                         [01;35m[K^[m[K
[01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1:[m[K [01;36m[Knote: [m[Kin expansion of macro ‘[01m[KINSTANTIATE_TEST_CASE_P[m[K’
   70 | [01;36m[KINSTANTIATE_TEST_CASE_P[m[K(
      | [01;36m[K^~~~~~~~~~~~~~~~~~~~~~~[m[K
In file included from [01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:62[m[K,
                 from [01m[K/home/joep/nav2_rolling_depends_ws/src/ros-simulation/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15[m[K:
[01m[K/home/joep/ros2_rolling/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16:[m[K [01;36m[Knote: [m[Kdeclared here
[... cut ...]
      |                 [01;31m[K~~~~~~~~~~^~~~[m[K
At global scope:
[01m[Kcc1plus:[m[K [01;35m[Kwarning: [m[Kunrecognized command line option ‘[01m[K-Wno-inconsistent-missing-override[m[K’
make[2]: *** [test/CMakeFiles/test_plugins.dir/build.make:63: test/CMakeFiles/test_plugins.dir/test_plugins.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:530: test/CMakeFiles/test_plugins.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
`

chapulina, 2020-11-04:
There's a `foxy` branch now. The `ros2` branch (Rolling) is being fixed in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1184. I'll close this, let's follow up on the build errors in https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1183/.

closed 2020-11-04 by chapulina