ros-simulation/gazebo_ros_pkgs #1191: Cannot spawn mesh data in XML file using Spawn_Entity.py

Drojas251, 2020-11-11:
**Description**
I am trying to spawn a robot model, described in a urdf, into gazebo 11 from a ros2 python launch file by using the spawn_entity.py node. I am publishing the xml data onto a topic with robot_state_publisher and reading that data from the topic with the spawn_entity node. I am able to spawn the robot successfully if the urdf is made up of built in geomotires ( box, cylinder..), however when I use a mesh to build my robot, spawn_entity node does now spawn those links with a mesh file in gazebo. I have tried multiple files types ( dae, stl, obj) and tried different urdf's.  

**To Reproduce**
I have used the launch file from gazebo_ros2_control_demos exactly. The only changes I made was just adding an additional link and joint to the urdf file that was used in that demo, and used a mesh for that link. I have provided a screen shot below. 
`

    xacro_file = os.path.join(gazebo_ros2_control_demos_path,
                              'urdf',
                              'test_cart_position.xacro.urdf')

    doc = xacro.parse(open(xacro_file))
    xacro.process_doc(doc)
    params = {'robot_description': doc.toxml()}

    node_robot_state_publisher = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        output='screen',
        parameters=[params]
    )

    spawn_entity = Node(package='gazebo_ros', executable='spawn_entity.py',
                        arguments=['-topic', 'robot_description',
                                   '-entity', 'cartpole'],
                        output='screen')

`

**Expected behavior**

I have made sure my files are in the shared folder and that everything is linked properly. If I can spawn a urdf with basic shapes, then I know that my launch file and the node is working as it should. However, I suspect that the mesh information is not completely being transferred from the robot_state_publisher topic to the spawn_entity node. 

**Screenshots**

Here is the portion of the URDF that includes the mesh. I have simply added this link and joint to the test_cart_position.xacro.urdf file in gazebo_ros2_control_demo. The result in gazebo does not display the mesh model. As I said before, I tried multiple file types. 

[image]

**Environment (please complete the following information):**

 - OS: [Ubuntu 20.04]
 - Gazebo version: [Gazebo 11]
 - ROS version: [foxy ]
  - gazebo_ros_pkgs: [e.g. debian install ]
 - glx server in use: output from `glxinfo | grep 'server glx vendor'` NVIDIA Corporation 

**Additional context**

I have tried the -database argument in the spawn_entity node and passed a gazebo model form the model data base, which is in sdf format and contains meshes, and that worked fine. However, publishing a urdf that contains meshes with robot stat publisher does not work for me.

newcanopies, 2021-05-03:
@Drojas251 I have this same issue - is URDF based on visual meshes supported in ROS2/Gazebo11 ?

newcanopies, 2021-05-04:
https://github.com/osrf/gazebo/issues/2992#issue-875697300

newcanopies, 2021-05-04:
https://github.com/osrf/gazebo/issues/2992

KACHAPPILLY2021, 2022-08-31:
I know its late but still, try this.
In mesh filename details, change 'package://' to absolute path on your system for every links.
Hope this works

munjer1990, 2023-11-09:
hi
Thanks a lot KACHAPPILLY2021. my case was  __ filename="package://my_robot_description/meshes/.....STL" 
I changed it to __ filename="$(find my_robot_description)/meshes/.....STL"  and meshes were appeared in Gazebo

ZAIDALI5, 2023-11-10:
@munjer1990 @KACHAPPILLY2021 

I still have the same issue, I tried to put absolute path, tried $(find package), but nothing working.
Ubuntu Humble 22.04
[spawn_entity.py-4] [INFO] [1699594524.611441036] [spawn_entity]: Waiting for entity xml on arduino_urdf
the package is called arduino_urdf, which inside has meshes folder and urdf folder and launch folder.

---------------------------------------------------------------------------------------------------------------
this is from arduinobot.urdf.xacro

<link name="base_link">
        <xacro:default_inertial mass="1.0"/>
        <visual>
            <origin rpy="0 0 0" xyz="-0.4755 -0.5 0"/>
            <geometry>
                <mesh filename="package://arduino_urdf/meshes/basement.STL" scale="0.01 0.01 0.01"/>
            </geometry>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="-0.4755 -0.5 0"/>
            <geometry>
                <mesh filename="package://arduino_urdf/meshes/basement.STL" scale="0.01 0.01 0.01"/>
            </geometry>
        </collision>
    </link>
    <joint name="virtual_joint" type="fixed">
        <parent link="world"/>
        <child link="base_link"/>
        <origin rpy="0 0 0" xyz="0 0 0"/>
    </joint>

---------------------------------------------------------------------------------------------------------------
this is full gazebo.launch.py

import os
from ament_index_python.packages import get_package_share_directory, get_package_prefix

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription, SetEnvironmentVariable
from launch.substitutions import Command, LaunchConfiguration
from launch.launch_description_sources import PythonLaunchDescriptionSource

from launch_ros.actions import Node
from launch_ros.parameter_descriptions import ParameterValue

def generate_launch_description():
    arduinobot_description_dir = get_package_share_directory('arduino_urdf')
    arduinobot_description_share = os.path.join(get_package_prefix('arduino_urdf'), 'share')
    gazebo_ros_dir = get_package_share_directory('gazebo_ros')

    model_arg = DeclareLaunchArgument(name='model', default_value=os.path.join(
                                        arduinobot_description_dir, 'urdf', 'arduinobot.urdf.xacro'
                                        ),
                                      description='Absolute path to robot urdf file'
    )

    env_var = SetEnvironmentVariable('GAZEBO_MODEL_PATH', arduinobot_description_share)

    robot_description = ParameterValue(Command(['xacro ', LaunchConfiguration('model')]),
                                       value_type=str)

    robot_state_publisher_node = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        parameters=[{'robot_description': robot_description}]
    )

    start_gazebo_server = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(gazebo_ros_dir, 'launch', 'gzserver.launch.py')
        )
    )

    start_gazebo_client = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(gazebo_ros_dir, 'launch', 'gzclient.launch.py')
        )
    )

    spawn_robot = Node(package='gazebo_ros', executable='spawn_entity.py',
                        arguments=['-topic', 'arduino_urdf',
                                   '-entity', 'arduinobot'
                                  ],
                        output='screen'
    )

    return LaunchDescription([
        env_var,
        model_arg,
        start_gazebo_server,
        start_gazebo_client,
        robot_state_publisher_node,
        spawn_robot
    ])

ZAIDALI5, 2023-11-10:
[image]

ZAIDALI5, 2023-11-10:
@munjer1990 @KACHAPPILLY2021 

I still have the same issue, I tried to put absolute path, tried $(find package), but nothing working.
Ubuntu Humble 22.04
[spawn_entity.py-4] [INFO] [1699594524.611441036] [spawn_entity]: Waiting for entity xml on arduino_urdf
the package is called arduino_urdf, which inside has meshes folder and urdf folder and launch folder.

---------------------------------------------------------------------------------------------------------------
this is from arduinobot.urdf.xacro

[image]

---------------------------------------------------------------------------------------------------------------
this is full gazebo.launch.py

import os
from ament_index_python.packages import get_package_share_directory, get_package_prefix

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription, SetEnvironmentVariable
from launch.substitutions import Command, LaunchConfiguration
from launch.launch_description_sources import PythonLaunchDescriptionSource

from launch_ros.actions import Node
from launch_ros.parameter_descriptions import ParameterValue

def generate_launch_description():
    arduinobot_description_dir = get_package_share_directory('arduino_urdf')
    arduinobot_description_share = os.path.join(get_package_prefix('arduino_urdf'), 'share')
    gazebo_ros_dir = get_package_share_directory('gazebo_ros')

    model_arg = DeclareLaunchArgument(name='model', default_value=os.path.join(
                                        arduinobot_description_dir, 'urdf', 'arduinobot.urdf.xacro'
                                        ),
                                      description='Absolute path to robot urdf file'
    )

    env_var = SetEnvironmentVariable('GAZEBO_MODEL_PATH', arduinobot_description_share)

    robot_description = ParameterValue(Command(['xacro ', LaunchConfiguration('model')]),
                                       value_type=str)

    robot_state_publisher_node = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        parameters=[{'robot_description': robot_description}]
    )

    start_gazebo_server = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(gazebo_ros_dir, 'launch', 'gzserver.launch.py')
        )
    )

    start_gazebo_client = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(gazebo_ros_dir, 'launch', 'gzclient.launch.py')
        )
    )

    spawn_robot = Node(package='gazebo_ros', executable='spawn_entity.py',
                        arguments=['-topic', 'arduino_urdf',
                                   '-entity', 'arduinobot'
                                  ],
                        output='screen'
    )

    return LaunchDescription([
        env_var,
        model_arg,
        start_gazebo_server,
        start_gazebo_client,
        robot_state_publisher_node,
        spawn_robot
    ])

RobinWolf, 2024-05-04:
Is there a fix yet? I suffer from the exact same problem!

rahgirrafi, 2024-06-02:
In the gazebo launch file, try adding positional arguments like this:

spawn_entity = Node(package='gazebo_ros', executable='spawn_entity.py',
                    arguments=['-topic', 'robot_description',
                                        '-entity', 'cartpole',
                                        "-x", "0.0", "-y", "0.0", "-z", "0.0"
],
                    output='screen')

2vin2vin, 2025-01-29:
I have been working with simulating linorobot2 and it worked earlier and out of nowhere today the model doesn't spawn.
I have checked even with turtlebot3 and the model doesn't spawn 
All of them worked earlier, without any changes made to the launch file now they don't
The issue should possibly be with gazebo 
ros2 launch linorobot2_gazebo gazebo_world.launch.py 
[INFO] [launch]: All log files can be found below /home/brahms
/.ros/log/2025-01-29-14-41-58-758884-D104714-37311
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [37313]
[INFO] [robot_state_publisher-2]: process started with pid [37315]
[INFO] [spawn_entity.py-3]: process started with pid [37317]
[INFO] [ekf_node-4]: process started with pid [37319]
[robot_state_publisher-2] [INFO] [1738141918.884631128] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-2] [INFO] [1738141918.884700091] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1738141918.884706924] [robot_state_publisher]: got segment camera_depth_link
[robot_state_publisher-2] [INFO] [1738141918.884710100] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-2] [INFO] [1738141918.884712908] [robot_state_publisher]: got segment front_left_wheel_link
[robot_state_publisher-2] [INFO] [1738141918.884715806] [robot_state_publisher]: got segment front_right_wheel_link
[robot_state_publisher-2] [INFO] [1738141918.884718445] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-2] [INFO] [1738141918.884721019] [robot_state_publisher]: got segment rear_left_wheel_link
[robot_state_publisher-2] [INFO] [1738141918.884723750] [robot_state_publisher]: got segment rear_right_wheel_link
[robot_state_publisher-2] [INFO] [1738141918.884726371] [robot_state_publisher]: got segment velodyne
[robot_state_publisher-2] [INFO] [1738141918.884729100] [robot_state_publisher]: got segment velodyne_base_link
[gazebo-1] Gazebo multi-robot simulator, version 11.10.2
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[spawn_entity.py-3] [INFO] [1738141919.105162694] [urdf_spawner]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1738141919.105332658] [urdf_spawner]: Loading entity published on topic robot_description
[spawn_entity.py-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[spawn_entity.py-3] [INFO] [1738141919.106057186] [urdf_spawner]: Waiting for entity xml on robot_description
[gazebo-1] Gazebo multi-robot simulator, version 11.10.2
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[gazebo-1] [Msg] Waiting for master.
[gazebo-1] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gazebo-1] [Msg] Publicized address: 192.168.16.129
[gazebo-1] [Msg] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[gazebo-1] [Err] [Scene.cc:227] Service call[/shadow_caster_material_name] timed out
[gazebo-1] [Err] [Scene.cc:249] Service call[/shadow_caster_render_back_faces] timed out
[gazebo-1] [Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[gazebo-1] [Wrn] [Scene.cc:463] Ignition transport [/scene_info] service call failed, falling back to gazebo transport [scene_info] request.

2vin2vin, 2025-01-29:
> I have been working with simulating linorobot2 and it worked earlier and out of nowhere today the model doesn't spawn. I have checked even with turtlebot3 and the model doesn't spawn All of them [...]

As the gazebo has launched without the robot, I tried to spawn the robot separately and got this error
ros2 run gazebo_ros spawn_entity.py -file /home/brahms/husky_ws/src/linorobot2/linorobot2_description/urdf/robots/4wd.urdf.xacro -entity linorobot2
[INFO] [1738147673.815682195] [spawn_entity]: Spawn Entity started
[INFO] [1738147673.815846110] [spawn_entity]: Loading entity XML from file /home/brahms/husky_ws/src/linorobot2/linorobot2_description/urdf/robots/4wd.urdf.xacro
[INFO] [1738147673.816098277] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[INFO] [1738147673.816226352] [spawn_entity]: Waiting for service /spawn_entity
[ERROR] [1738147703.861967413] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[ERROR] [1738147703.862745864] [spawn_entity]: Spawn service failed. Exiting.
[ros2run]: Process exited with failure 1

Then i tried out 
ros2 topic list
/parameter_events
/rosout

Gazebo is still open, it seems like gazebo did not start with ros
but in my launch file I am starting it with
ExecuteProcess(
            cmd=['gazebo', '--verbose', '-s', 'libgazebo_ros_factory.so',  '-s', 'libgazebo_ros_init.so', LaunchConfiguration('world')],
            output='screen'
        ),

I am really confused on how to get this working