ros-simulation/gazebo_ros_pkgs #1195: 👨‍🌾 ros2_gazebo_pkgs-install_pkg failing on Rolling

azeey, 2020-11-18:
`ros2_gazebo_pkgs-install_pkg_rolling-focal-amd64` [fails](https://build.osrfoundation.org/job/ros2_gazebo_pkgs-install_pkg_rolling-focal-amd64/40/) and has been failing for the last 9 builds with the following error log (with color codes removed)

```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.17.0.2
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.17.0.2
[Wrn] [GuiIface.cc:297] Couldn't locate specified .ini. Creating file at "/root/.gazebo/gui.ini"
[INFO] [1605713970.815893060] [gazebo_ros_node]: ROS was initialized without arguments.
malloc(): invalid size (unsorted)

```

sloretz, 2020-11-25:
I can't reproduce with the latest `ros2` sources. I think this just needs a new bloom release into ROS rolling. The [Rolling binary job for gazebo_ros](http://build.ros2.org/job/Rbin_uF64__gazebo_ros__ubuntu_focal_amd64__binary/70/) has been failing for a while, so it's likely that there was an ABI break upstream (expect in Rolling) and the package hasn't been able to be rebuilt yet.

http://build.ros2.org/job/Rbin_uF64__gazebo_ros__ubuntu_focal_amd64__binary/70/

sloretz, 2020-11-25:
https://github.com/ros/rosdistro/pull/27493

chapulina, 2020-11-30:
It's fixed :sparkles:

closed 2020-11-30 by chapulina