ros-simulation/gazebo_ros_pkgs #1211: Increase /clock frequency

iluetkeb, 2021-01-13:
**Description**

The current default frequency for /clock is 10Hz, which gives 100ms clock resolution for the "now()" call. This is *very* coarse and represents a huge difference to most real-world robots, where the usable clock precision is in the microsecond range, and also to ROS1, where it used to be 1ms, IIRC. Since all nodes need the sim_time, increasing the frequency has a cost, of course, therefore a balance needs to be found between precision and message sending overhead.

There is a parameter for changing this, but I would say that the default should be reasonable, and in my considered opinion, 10Hz is not reasonable for the reasons outlined above.

**Expected behavior**

I would suggest to increase the default frequency to 1000, to mimick ROS1 behavior.

**Environment (please complete the following information):**

  - gazebo_ros_pkgs: up to master

musamarcusso, 2021-01-15:
Trying the instructions in the wiki also seem to not work.
I created a `params.yaml` file 
```yaml
gazebo:
    ros__parameters:
        publish_rate: 50.0
```
and called as instructed [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time). Gazebo seems to think the file is a `world` file.

```
gazebo --verbose -s libgazebo_ros_init.so __params:=params.yaml
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:178] Failed to load plugin libgazebo_ros_init.so: libgazebo_ros_init.so: cannot open shared object file: No such file or directory
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.178.25
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.178.25
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [__params:=params.yaml]
[Err] [Server.cc:444] Could not open file[__params:=params.yaml]
[Wrn] [Server.cc:359] Falling back on worlds/empty.world
[Msg] Loading world file [/usr/share/gazebo-11/worlds/empty.world]
^C[Wrn] [Publisher.cc:136] Queue limit reached for topic /gazebo/default/physics/contacts, deleting message. This warning is printed only once.
```

iluetkeb, 2021-01-15:
btw, I should add that we couldn't figure out a way to actually _use_ the parameter "update_rate". There doesn't seem to be a way to pass it to the plugin.

ivanpauno, 2021-01-22:
The "update_rate" is being throttled, there's another parameter called "publish_rate": https://github.com/ros-simulation/gazebo_ros_pkgs/blob/204abd4f03e0759115eee68031111f132d42724a/gazebo_ros/src/gazebo_ros_init.cpp#L174-L176
Did you try that one?

The default is [10Hz](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/204abd4f03e0759115eee68031111f132d42724a/gazebo_ros/src/gazebo_ros_init.cpp#L135), modifying it sounds ok to me.

This sounds like a lack of docs problem to me, I'm not sure where to document this though...

tim-fan, 2021-01-25:
Hello, just to add my comments because I am having similar issues (using ros foxy).

Regarding the use of the `publish_rate` param, I've tried setting that following [this guide](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time#clock-rate), without success. (i.e `ros2 topic hz /clock` shows 10 Hz, regardless of `publish_rate` setting.

BTW I'd say if clock rate is intended to be configurable, it would be nice to expose it as a launch arg in `gazebo.launch.py` and `gzserver.launch.py`.

The inability to set the clock rate is causing problems for me, because as far as I can tell, the clock publish rate limits the rate at which `rclpy` publishers can spin. For instance, when running [examples_rclpy_minimal_publisher/publisher_local_function.py](https://github.com/ros2/examples/blob/master/rclpy/topics/minimal_publisher/examples_rclpy_minimal_publisher/publisher_local_function.py) unmodified and with `use_sim_time:=true`, the publisher publishes at 2Hz. But if I set [timer period](https://github.com/ros2/examples/blob/master/rclpy/topics/minimal_publisher/examples_rclpy_minimal_publisher/publisher_local_function.py#L36) to `0.05` for a 20Hz loop rate, the publisher only publishes at 10Hz (the gazebo clock publish rate). 

I'm not sure if this limiting of spin rate is expected behavior, or an `rclpy` bug. In any case I have a node that I'd like to run at 50Hz, and because I can't figure out how to configure the gazebo clock publish rate, my only solution atm is to change the default clock rate in `gazebo_ros` source and recompile.

iluetkeb, 2021-01-25:
[publish_rate]
> Did you try that one?

We tried that one as well, but the issue is that we cannot pass ROS parameters on the Gazebo command line. For other plugins, their parameters are given in the SDF (when the plugin is used for some model part), but this doesn't work for initial parameters.

ivanpauno, 2021-01-25:
> But if I set timer period to 0.05 for a 20Hz loop rate, the publisher only publishes at 10Hz (the gazebo clock publish rate).

That doesn't seem related to gazebo, just `rclpy` not being able to track the timer period correctly.

joshnewans, 2021-03-05:
@tim-fan Were you able to find any workaround?
I aslo have had no success setting the `publish_rate` as per that guide (i.e. creating a params file and executing `gazebo -s libgazebo_ros_init.so __params:=/tmp/params.yaml`.

I also agree that this is something that should be exposed by the launch files.

tim-fan, 2021-03-05:
Hi @joshnewans, I was not able to set clock rate via params either (maybe this should be listed as a separate issue).

For now I'm building the #1214 branch from source, until that gets merged/released.

jacobperron, 2022-05-17:
https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1391 has made it more convenient to set ROS parameters (including `publish_rate`) from the launch file. The wiki has also been updated with the commands to set parameters from the command-line; the `__params:=` syntax is outdated.

PR #1391 (2022-05-17): Expose launch parameter for `params_file`
This PR exposes a `params_file` launch argument for `gazebo` with which the clock rate (or any other ROS parameters) can be set.

For example:

``` yaml
# params.yml
gazebo:
    ros__parameters:
        publish_rate: 400.0
```

`ros2 launch gazebo_ros gazebo.launch.py params_file:=params.yml`

See [wiki](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration%3A-ROS-Clocks-and-sim-time) for more

- Closes #1305 
- Closes #1211 
- Closes #1247