ros-simulation/gazebo_ros_pkgs #1216: Problem finding googletest INSTANTIATE_TEST_SUITE_P macro when building on MacOS with ROS2 Foxy

jpswensen, 2021-01-27:
Describe your issue here and explain how to reproduce it.

**Description**

Trying to build gazebo_ros_pkgs on MacOS. Following the instructions at http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros after having followed the instructions at https://index.ros.org/doc/ros2/Installation/Crystal/macOS-Development-Setup/ to get Foxy build from sources.

It appears that the issue ended up being a problem with not finding macros associated with googletest and the test suite of gazebo_ros_pkgs. In particular, it seems to be choking on  INSTANTIATE_TEST_SUITE_P.

The first instance of this that throws an error is in gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp

```
INSTANTIATE_TEST_SUITE_P(
  Plugins, TestPlugins, ::testing::Values(
    TestParams({{"-s", "./libargs_init.so"}, {"test"}}),
...
```
The full  compiler output surrounding the error is

```
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:71:3: error: unknown type name 'Plugins'
  Plugins, TestPlugins, ::testing::Values(
  ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:71:23: error: parameter type 'TestPlugins' is an abstract class
  Plugins, TestPlugins, ::testing::Values(
                      ^
/Users/jpswensen/ros2-foxy/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:489:16: note: unimplemented pure virtual method 'TestBody' in 'TestPlugins'
  virtual void TestBody() = 0;
               ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:71:36: error: no type named 'Values' in namespace 'testing'
  Plugins, TestPlugins, ::testing::Values(
                        ~~~~~~~~~~~^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:72:16: error: expected ')'
    TestParams({{"-s", "./libargs_init.so"}, {"test"}}),
               ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:72:15: note: to match this '('
    TestParams({{"-s", "./libargs_init.so"}, {"test"}}),
              ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:73:16: error: expected ')'
    TestParams({{"-s", "./libcreate_node_without_init.so"}, {"test"}}),
               ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:73:15: note: to match this '('
    TestParams({{"-s", "./libcreate_node_without_init.so"}, {"test"}}),
              ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:74:16: error: expected ')'
    TestParams({{"-s", "./libmultiple_nodes.so"}, {"testA", "testB"}}),
               ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:74:15: note: to match this '('
    TestParams({{"-s", "./libmultiple_nodes.so"}, {"testA", "testB"}}),
              ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:76:7: error: expected ')'
      {{"-s", "libgazebo_ros_init.so", "worlds/ros_world_plugin.world",
      ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:75:15: note: to match this '('
    TestParams(
              ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:79:7: error: expected ')'
      {{"-s", "libgazebo_ros_init.so", "-s", "libgazebo_ros_factory.so",
      ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:78:15: note: to match this '('
    TestParams(
              ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:81:16: error: expected ')'
    TestParams({{"-s", "libgazebo_ros_init.so", "worlds/sdf_node_plugin.world"}, {"/foo/my_topic"}})
               ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:81:15: note: to match this '('
    TestParams({{"-s", "libgazebo_ros_init.so", "worlds/sdf_node_plugin.world"}, {"/foo/my_topic"}})
              ^
/Users/jpswensen/ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1: error: C++ requires a type specifier for all declarations
INSTANTIATE_TEST_SUITE_P(
```
I did attempt to go find a solution at answers.gazebosim.org, but there seemed to be very few questions about building and even fewer about building on MacOS, so decided to post an issue here, instead of there.

I suspect there is some mismatch between the branch checkout I should be using. I am using the ros2 branch in the step given of 

`vcs custom --args checkout ros2`

I also tried "foxy" instead of "ros2" in that command, but it didn't work.

I also went back and tried to recursively search for INSTANTIATE_TEST_SUITE_P in my ros2 source/build directory (including the googletest subfolder) and in the gazebo_ros_pkg source/build directory, and couldn't find a definition for that macro in any header or source files.

**To Reproduce**

Steps to reproduce the behavior:
1. Install Foxy version of ROS2 following the instructions at https://index.ros.org/doc/ros2/Installation/Crystal/macOS-Development-Setup/
2. Follow the instructions at http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros to try and build these packages.

**Expected behavior**

I would expect compilation to find the INSTANTIATE_TEST_SUITE_P macro and create the test in the the test suite accordingly.

**Environment (please complete the following information):**

 - OS: MacOS using homebrew for most of the dependencies
 - Gazebo version: [e.g. Gazebo 9]
 - ROS version: Foxy
  - gazebo_ros_pkgs: source install using the link above to pull packages

**Additional context**

Add any other context here.

jpswensen, 2021-01-27:
I guess I can amend this and close it. I ended up getting it to work, but here are the additional step I had to take.

1. For some reason, the command `vcs custom --args checkout ros2` did not work,
2. However, copying the gazebo_ros_pkgs.repos file from the Foxy branch and then performing the `vcs import src < gazebo_ros_pkgs.repos` did work. After inspection, the only different is that the version for gazebo_ros_pkgs is "foxy" and the other two are ros2.

closed 2021-01-27 by jpswensen

scpeters, 2021-07-13:
I recently experienced this problem and I believe it was related to sourcing the wrong setup file (foxy, when I meant to only use galactic)