ros-simulation/gazebo_ros_pkgs #1234: ROS time is not respected in gazebo plugins

jacobperron, 2021-02-17:
**Description**

If ROS time is set to track a clock different from the one provided by Gazebo, then the time stamps of ROS messages coming from Gazebo ROS plugins do not agree with the rest of the ROS system.

**To Reproduce**

The easiest way to reproduce the behavior is to launch a ROS system with the parameter `use_sim_time:=false`.
Nodes will report timestamps wrt wall time, but topics from gazebo_ros will continue to use sim time.

It seems to get more confusing in ROS 1, where some topics from gazebo_ros use sim time and others use ROS time.

**Expected behavior**

TBH, I'm not sure what the expected behaviour should be, but it seems like we should at least be consistent and provide recommendations for others creating their own plugins.

In particular, publishers to `/tf` are using ROS time in ROS 1. However, because we're using simulation time in ROS 2, I notice a broken TF tree.

On one hand, using simulation time everywhere makes sense since we can get a more accurate timestamp for the data. But, this makes using gazebo_ros difficult in cases where we don't want to use sim time.

**Additional context**

Here are some places where simulation time is used for ROS messages stamps:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/50c2d1b329936b2c60cdd78aabbf616ff691e43c/gazebo_plugins/src/gazebo_ros_p3d.cpp#L203

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/50c2d1b329936b2c60cdd78aabbf616ff691e43c/gazebo_plugins/src/gazebo_ros_tricycle_drive.cpp#L412-L413

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/50c2d1b329936b2c60cdd78aabbf616ff691e43c/gazebo_plugins/src/gazebo_ros_ft_sensor.cpp#L206

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/50c2d1b329936b2c60cdd78aabbf616ff691e43c/gazebo_plugins/src/gazebo_ros_joint_state_publisher.cpp#L157

*ROS 1 equivalents*

It seems an arbitrary choice in ROS 1 if ROS message headers are populated with simulation time or ROS time.

The following use sim time:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/b0ed38f9ecedbe929340f5e8b0aa7a457248e015/gazebo_plugins/src/gazebo_ros_p3d.cpp#L225

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/b0ed38f9ecedbe929340f5e8b0aa7a457248e015/gazebo_plugins/src/gazebo_ros_ft_sensor.cpp#L204-L205

These others use ROS time:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/b0ed38f9ecedbe929340f5e8b0aa7a457248e015/gazebo_plugins/src/gazebo_ros_tricycle_drive.cpp#L168

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/b0ed38f9ecedbe929340f5e8b0aa7a457248e015/gazebo_plugins/src/gazebo_ros_joint_state_publisher.cpp#L148

adityapande-1995, 2021-07-19:
The timestamps on sensor and transform messages are currently associated with the simulation time, i.e. gazebo's solver. Since the real time factor can vary based on system load, attaching system time (or any non-simulation time) to the messages would lead to messages being published at variable time intervals. For e.g, in simulation, odometry may reflect that your vehicle is running at a constant speed, but if the messages are published according to system time, one may conclude that the vehicle is running at variable speed.

Moreover, in gazebo classic, the sensors and physics run on separate threads, which are not synchronized. This can lead to situations where the physics and sensors update with the same "sim time stamp" but are published at different "wall time stamps" which can mess up SLAM algorithms. 

It is recommended to to use only simulation time for these plugins, however if needed, an intermediate node can be added that changes the timestamp on the required messages. The relevant documentation and warnings will be added soon.

jacobperron, 2021-07-19:
Thanks @adityapande-1995

To summarize our offline discussion, we'll take the following actions:

1. Ensure all outgoing data from gazebo_ros is stamped with Gazebo time
2. Document that timestamps will always be Gazebo time (and ignore the ROS parameter `use_sim_time`).
3. Emit a runtime warning if a user sets a gazebo_ros node's `use_sim_time` parameter to `false`.

adityapande-1995, 2021-07-19:
Relevant (pull request)[https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1298]

jacobperron, 2021-08-09:
Now all ROS 2 plugins always use simulation time, regardless of what `use_sim_time` is set to. Thanks @adityapande-1995 :bow:

closed 2021-08-09 by jacobperron