ros-simulation/gazebo_ros_pkgs #1236: [ROS2] Could not spawn URDF model

atinfinity, 2021-02-18:
**Description**

I tried to spawn URDF model using gazebo_ros.
But, I could not spawn URDF model.

**To Reproduce**

I used <https://github.com/ros/urdf_tutorial/blob/master/urdf/05-visual.urdf>.

```
ros2 launch gazebo_ros gazebo.launch.py
ros2 run gazebo_ros spawn_entity.py -file $HOME/dev_ws/src/urdf_tutorial/urdf/05-visual.urdf -entity 05
```

The message of this command is as follows.

```
$ ros2 run gazebo_ros spawn_entity.py -file $HOME/dev_ws/src/urdf_tutorial/urdf/05-visual.urdf -entity 05
[INFO] [1613667021.959447709] [spawn_entity]: Spawn Entity started
[INFO] [1613667021.960142006] [spawn_entity]: Loading entity XML from file /home/dandelion/dev_ws/src/urdf_tutorial/urdf/05-visual.urdf
[INFO] [1613667021.962618085] [spawn_entity]: Waiting for service /spawn_entity, timeout = 5
[INFO] [1613667021.963103341] [spawn_entity]: Waiting for service /spawn_entity
[INFO] [1613667021.967460842] [spawn_entity]: Calling service /spawn_entity
[INFO] [1613667022.131465118] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [05]
```

**Expected behavior**

I expected that `05-visual.urdf` spawns in Gazebo world.

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

 - OS: Ubuntu 20.04
 - Gazebo version: Gazebo 11.3.0
 - ROS version: ROS2 foxy
  - gazebo_ros_pkgs: debian install
 - glx server in use: output from `glxinfo | grep 'server glx vendor'` (glxinfo is in mesa-utils package for Debian/Ubuntu)

**Additional context**

I found related issue.

- https://github.com/ros2/demos/pull/349
- https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1191

newcanopies, 2021-05-03:
I have this same issue on the same Environment.

newcanopies, 2021-05-04:
https://github.com/osrf/gazebo/issues/2992#issue-875697300

newcanopies, 2021-05-04:
https://github.com/osrf/gazebo/issues/2992

atinfinity, 2021-05-04:
@newcanopies Thank you for your information. I converted from URDF to SDF to avoid this issue.

newcanopies, 2021-05-04:
@atinfinity SDF is not compatible with moveit2, 
or were you able to run kinematic controller plugins in the SDF model?

atinfinity, 2021-05-05:
@newcanopies I don't use moveit2. So, I used model file properly as follows.

- sdf: spawn model to Gazebo
  - https://github.com/atinfinity/megarover_samples_ros2/blob/main/launch/vmegarover_with_sample_world.launch.py
- xacro or URDF: robot_state_publisher
  - https://github.com/atinfinity/megarover_samples_ros2/blob/main/launch/robot_state_publisher.launch.py

aihoque2, 2023-02-21:
check your `~/.gazebo/client-xxxx` for a `default.log` file. it should contain your issue. What helped me was moving my package into `~/.gazebo/models/` directory, and I wonder how we do that in the ros2 launch file.