ros-simulation/gazebo_ros_pkgs #1242: gazebo_ros_multicamera plugin: changing camera names in .sdf file breaks the plugin

canberkgurel, 2021-02-27:
**Description**

I have been modifying the [stereo_camera.sdf](https://github.com/PX4/PX4-SITL_gazebo/blob/bebb9a95f0b61bf9e4c3de345fab70985c1329b3/models/stereo_camera/stereo_camera.sdf) to create a simulation model of my stereo-pair camera. To keep the rest of the nodes identical with the real hardware I need to output the camera images and information on the same ROS topics used by the real hardware drivers. The problem is that when I rename the camera names "right" and "left" as "fisheye1" and "fisheye2", respectively, I noticed that the plugin breaks, i.e., not outputting images nor the camera information. This is a silent failure, the compiler doesn't throw any errors. 

**To Reproduce**

Steps to reproduce the behavior:

When I use the below configuration everything works as expected. See Screenshot1.

         <camera name="left">
          <pose>0.1 0.035 0 0 0 0</pose>
          <horizontal_fov>1.953332298</horizontal_fov>
          <image>
            <width>848</width>
            <height>800</height>
            <format>R8G8B8</format>
          </image>
          <clip>
            <near>0.02</near>
            <far>300</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.007</stddev>
          </noise>
        </camera>
        <camera name="right">
          <pose>0.1 -0.035 0 0 0 0</pose>
          <horizontal_fov>1.953332298</horizontal_fov>
          <image>
            <width>848</width>
            <height>800</height>
            <format>R8G8B8</format>
          </image>
          <clip>
            <near>0.02</near>
            <far>300</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.007</stddev>
          </noise>
        </camera>

When I modify the camera names, the plugin no longer works. See Screenshot2.

        <camera name="fisheye2">
          <pose>0.1 0.035 0 0 0 0</pose>
          <horizontal_fov>1.953332298</horizontal_fov>
          <image>
            <width>848</width>
            <height>800</height>
            <format>R8G8B8</format>
          </image>
          <clip>
            <near>0.02</near>
            <far>300</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.007</stddev>
          </noise>
        </camera>
        <camera name="fisheye1">
          <pose>0.1 -0.035 0 0 0 0</pose>
          <horizontal_fov>1.953332298</horizontal_fov>
          <image>
            <width>848</width>
            <height>800</height>
            <format>R8G8B8</format>
          </image>
          <clip>
            <near>0.02</near>
            <far>300</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.007</stddev>
          </noise>
        </camera>

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**

Screenshot 1:

[image]

Screenshot 2:

[image]

**Environment (please complete the following information):**

 - OS: Ubuntu 18.04.5 LTS
 - Gazebo version: Gazebo 9.16.0
 - ROS version: Melodic
  - gazebo_ros_pkgs: How do I find out?
 - glx server in use: server glx vendor string: NVIDIA Corporation

**Additional context**

Add any other context here.

canberkgurel, 2021-03-11:
@codebot @davetcoleman @jbohren @nkoenig @scpeters @tfoote @wjwwood Hi All, any ideas on this problem? Thanks!

codebot, 2021-03-11:
Looks like the camera names are currently hard-coded in the multicamera plugin:
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_multicamera.cpp#L80
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_multicamera.cpp#L85

You could modify the plugin to extract parameters from the SDF for the camera names.

canberkgurel, 2021-03-11:
@codebot Thanks for the pointers! I can perhaps make them configurable and submit a PR!