ros-simulation/gazebo_ros_pkgs #1247: ROS2 plugin Cannot set parameters (e.g. publish_rate)

joshnewans, 2021-03-06:
**Description**

The ROS2 plugin for Gazebo (`libgazebo_ros_init`?) does not seem to accept parameters, which is required to change the clock frequency.
There is an existing issue, #1211, discussing increasing the default clock frequency, and in that thread myself and another user (tim-fan) noted that regardless of the default, the `publish_rate` parameter cannot be set/changed despite [this guide](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time#clock-rate) indicating that it should be possible.

**To Reproduce**

Steps to reproduce the behavior:
Follow the guide at https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time#clock-rate
1. Create a YAML file with the appropriate parameters
2. Launch gazebo with the params file set `gazebo -s libgazebo_ros_init.so __params:=/tmp/params.yaml`
3. Check `ros2 param get /gazebo publish_rate` or `ros2 topic hz /clock`

**Expected behavior**

The `publish_rate` and `/clock` rate should match whatever is set in the params file. Instead it is always 10Hz.

**Environment (please complete the following information):**

 - OS: Foxy Docker (Ubuntu 20.04)
 - Gazebo version: Gazebo 11.3
 - ROS version: Foxy
  - gazebo_ros_pkgs: Debian install
 - glx server in use: NVIDIA Corporation

**Additional context**

This seems to be where the parameter is processed, I'm just not sure how to direct it there.
Also, as mentioned in those threads, this parameter should be exposed by the launch files, but I guess that can be a separate issue once this is resolved.

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/8a320668a5a6c84242547e13e04dbd5e1029c063/gazebo_ros/src/gazebo_ros_init.cpp#L167-L171

allenh1, 2021-11-09:
Is there any update on this? I'm stuck with this at the moment, and need to change this in a launch file to 500 Hz.

jacobperron, 2022-05-17:
`gazebo -s libgazebo_ros_init.so __params:=/tmp/params.yaml` is outdated syntax. We've updated the [wiki](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time) to reflect the new syntax for ROS arguments: `gazebo -s libgazebo_ros_init.so --ros-args --params-file params.yaml`

jacobperron, 2022-05-17:
https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1391 has also made it more convenient to set ROS parameters from the launch file.

PR #1391 (2022-05-17): Expose launch parameter for `params_file`
This PR exposes a `params_file` launch argument for `gazebo` with which the clock rate (or any other ROS parameters) can be set.

For example:

``` yaml
# params.yml
gazebo:
    ros__parameters:
        publish_rate: 400.0
```

`ros2 launch gazebo_ros gazebo.launch.py params_file:=params.yml`

See [wiki](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration%3A-ROS-Clocks-and-sim-time) for more

- Closes #1305 
- Closes #1211 
- Closes #1247
closed 2022-05-17 by jacobperron