ros-simulation/gazebo_ros_pkgs #1268: Feature: Support for one-step simulation/control of ROS-Gazebo (that is, ROS-Gazebo to be one-stepped synchronously)

safdark, 2021-04-26:
**Description**
This is a feature request, with proposed [draft commit](https://github.com/safdark/gazebo_ros_pkgs/commit/4d1d3cd74cbcc7a6331247f914adde8df5f7d6c2), for supporting one-step control of Gazebo, via ROS. Gazebo independently offers one-step control via the gazebo::msgs::WorldControl message, but that functionality is not presently exposed to ROS nodes via the gazebo_ros_api plugin. This feature request is intended to address that gap.

**Motivation**:
This is mostly in the context of ROS-based machine-learning systems (specifically reinforcement learning, but applicable generally to ML as well). In the case of RL, AI agents must be able to associate actions with their consequent observations and rewards, as part of the reward-based feedback loop. However, since Gazebo and ROS (indeed all nodes within ROS) operate asynchronously with respect to each other, it is presently not possible to draw time-synchronized or time-quantized action-state-reward associations, which impacts both training and inference stages of the RL system. A 'one-step control' feature that can be invoked directly by ROS nodes (or scripts/actions/services) would address this need. An ML agent (running as a ROS node, for example) will then be able to run training/inference on cleanly synchronized and quantized action-state-reward tuples generated by the system across the given 'step'.

**Proposed fix**

Specifics:
- gazebo_ros_api_plugin.h exposes a new ROS-facing service ('step_control') with corresponding [StepControl.srv](https://raw.githubusercontent.com/safdark/gazebo_ros_pkgs/daryl_glue_dev/gazebo_msgs/srv/StepControl.srv) message.
- This service in-turn invokes the WorldControl functionality in gazebo that pauses and one-steps the gazebo simulator.
- Effective one-step size (in seconds) is determined by the 'max_step_size' parameter configured for the gazebo world.

Prerequisites:
- "/use_sim_time" ROS parameter must be set to True

Please see [draft commit](https://github.com/safdark/gazebo_ros_pkgs/commit/4d1d3cd74cbcc7a6331247f914adde8df5f7d6c2).

**Considerations**
I have verified (and have been using this [commit](https://github.com/safdark/gazebo_ros_pkgs/commit/4d1d3cd74cbcc7a6331247f914adde8df5f7d6c2)) in my own project, so am happy to polish it up and submit a pull request for it. However, there are some design considerations that I'm hoping to get feedback from the community about, before I embark on that project:

Issues and proposed solutions:
- _Risk of deadlock between ROS and Gazebo during one-step simulation_
_Cause_: ROS depends on Gazebo for its /clock (use_sim_time=True). This feature will make Gazebo depend on ROS to issue one-step calls as well. So, any sleep() invocations on ros.Rate() or ros.Duration() will block until the next one-step invocation.
_Precedent_: The same risk exists with 'pause_physics/unpause_physics' already, so it can be argued that this is a FOL.
_Solution_: Alternatively, a one-step timeout could be implemented that deactivates the one-step control (unpauses Gazebo) if it is not one-stepped within a certain timeout (say 1 sec?). This could be a viable solution because most one-step drivers would be issuing steps at a higher frequency, and would unlikely be pushing the limit of this timeout anyway.

Open to feedback about other considerations.

**Environment (please complete the following information):**
All applicable.

safdark, 2021-04-26:
Issue was brought up earlier in 2016 (https://github.com/ros-simulation/gazebo_ros_pkgs/issues/489).
Made the request a bit more formal with this one.

safdark, 2021-05-10:
If this is an acceptable feature, checking it into ROS1 might allow it to be picked up automatically during the pending ROS2 port of gazebo_ros_api_plugin.hpp/cpp as well.

safdark, 2021-05-13:
For anyone who would like to see this feature, please upvote on the original description above, or add your thoughts via a comment.

safdark, 2021-05-31:
Feature [implementation](https://github.com/safdark/gazebo_ros_pkgs/tree/issue_1268). Will submit a pull request when ready.

safdark, 2021-06-04:
Submitted [PR](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1280) for feature.

Kettenhoax, 2021-08-24:
I have a similar issue, where I try to compare the simulated pose and velocity of a vehicle with recorded poses and velocities. The original vehicle commands (for instance Twist messages) are in the same recording, and I could align the response by timestamp if Gazebo would operate in the same time frame as the recorded ROS(2) bag. My primary goal is to verify the simulation accuracy.

In contrast to the suggested implementation, the most convenient way for me to accomplish this would be gazebo_ros to subscribe to the `/clock` topic published by `ros2 bag play`, and to step the simulation accordingly. I realize that the resolution of the published clock increments would have to be forced to a multiple of Gazebo's max_step_size.

Even though this approach feels out of place in this issue with its current wording, in essence my use case still requires some kind of external simulation stepping. I just wanted to add additional motivation to this issue.

safdark, 2021-08-25:
Thanks for the comment, @Kettenhoax!
Your suggestion is *not* out of place for this issue. In fact, I second it.
A feature for gazebo to respect (subscribe to) the /clock topic, controlled via a new gazebo-specific param (such as "subscribe_to_clock" or "time_slave"), would work for any situation requiring external control of the system clock (including my use case). The only complication, as you mentioned, is with the appropriate behavior for gazebo, when /clock messages fall within the max_step_size boundary. We'll have to look to the community for suggestions about that...

Meanwhile, absent the ideal solution, the [PR](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1280) that I submitted could potentially work for your use-case as well (with a small hack) if you are blocked, and if your analysis is not sensitive to the max_step_size boundary. Let me know if you'd like me to elaborate.

safdark, 2021-10-30:
... Adding to my previous answer, @soldierofhell : If *use_sim_time*=*False* (i.e, you were using ROS as the time authority), then there would be no deadlock, since ROS would be the time authority, and gazebo_ros would not publish to /clock at all. And, you'd need to utilize step_control() specifically to keep Gazebo in sync with ROS. Whether that works for you is more a question of whether you want ROS or Gazebo to be the time authority.

goekce, 2022-02-25:
For my robotics class I was playing with [a RL package using ROS2 and Turtlebot3](https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning). Probably because Gazebo time cannot be controlled exactly (in other words does not allow step control), the training node waits for a predefined time like 100 ms at each iteration. If the simulation computer is slow, the training step may take longer than 100 ms and the simulation may advance more than expected. This results in unpredictable results and breaks the training.

My plan was to adapt the code from the [Tensorflow example for deep Q learning](https://colab.research.google.com/github/tensorflow/agents/blob/master/docs/tutorials/1_dqn_tutorial.ipynb), then I noticed that Gazebo needs a step simulator command and this is how I arrived here.

I expected more attention to this issue. Is there another way that to do step control?

Thank you @safdark for pushing this issue and implementing a PR. Unfortunately I need a ROS2 solution.

safdark, 2022-02-25:
Thanks for the comment, @goekce. I too will soon need to port over the solution (or smth similar) for ROS2 in my own setup. I'll probably get some time in late March (or April) to look into this further and will incorporate the use cases and implementation ideas listed by commenters in this thread at that time.

Are you able to make do without this until then?

goekce, 2022-02-25:
I will try the following:

- use more computing resources
- increase the iteration time
- integrate `un/pause_physics()`

Glad for your concern @safdark :slightly_smiling_face:, in worst case I can show other examples to my students.