ros-simulation/gazebo_ros_pkgs #1272: spawn_entity.py cannot publish URDF meshes to Gazebo11

newcanopies, 2021-05-04:
Missing ur10.urdf.xacro model
[image]
Missing ur10 meshes
[image]

ur10.urdf.xacro https://gist.github.com/newcanopies/bb0206fd35c3e8668a93dac0b038ae05

ur10_robot_state_publisher_gazebo.launch 

```
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

def generate_launch_description():

  use_sim_time = LaunchConfiguration('use_sim_time', default='false')
  urdf_file_name = 'models/ur10/ur10.urdf.xacro'

  print("urdf_file_name : {}".format(urdf_file_name))

  urdf = os.path.join(
      get_package_share_directory('ur10_gazebo'),
      urdf_file_name)

  return LaunchDescription([

        DeclareLaunchArgument(
            'use_sim_time',
            default_value='false',
            description='Use simulation (Gazebo) clock if true'),
        Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            name='robot_state_publisher',
            output='screen',
            parameters=[{'use_sim_time': use_sim_time}],
            arguments=[urdf])
  ])
```

# TO REPRODUCE

### terminal 1:
either use command:
 $ `gazebo --verbose -s libgazebo_ros_factory.so`
or use command:
$ `ros2 launch gazebo_ros gazebo.launch.py`

```
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ur10_gazebo/models/ur10/meshes/dae/wrist3.dae]
[Err] [MeshShape.cc:64] Failed to find mesh file [model://ur10_gazebo/models/ur10/meshes/dae/wrist3.dae]
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_ros_control.so: 
libgazebo_ros_control.so: cannot open shared object file: No such file or directory
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ur10_gazebo/models/ur10/meshes/dae/upperarm.dae]
```

I also tried including the meshes with: 
< mesh filename="file://$(find ur10)/models/ur10/meshes/stl/upperarm.stl" />

```
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [$(find ur10)/models/ur10/meshes/stl/upperarm.stl]

```
as well as using dae versions of the meshes:
<mesh filename="**file**://**$(find ur10)**/models/ur10/meshes/dae/wrist1.dae"/>
returning same error as above.

### terminal 2:
$ `ros2 launch ur10_gazebo ur10_robot_state_publisher_gazebo.launch .py gdb:=true`

```
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : models/ur10/ur10.urdf.xacro
[INFO] [robot_state_publisher-1]: process started with pid [640]
[robot_state_publisher-1] [WARN] [1620150401.140379100] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 2 children
[robot_state_publisher-1] Link base had 0 children
[robot_state_publisher-1] Link shoulder_link had 1 children
[robot_state_publisher-1] Link upper_arm_link had 1 children
[robot_state_publisher-1] Link forearm_link had 1 children
[robot_state_publisher-1] Link wrist_1_link had 1 children
[robot_state_publisher-1] Link wrist_2_link had 1 children
[robot_state_publisher-1] Link wrist_3_link had 2 children
[robot_state_publisher-1] Link ee_link had 0 children
[robot_state_publisher-1] Link tool0 had 0 children
[robot_state_publisher-1] [INFO] [1620150401.146911300] [robot_state_publisher]: got segment base
[robot_state_publisher-1] [INFO] [1620150401.147041500] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1620150401.147055900] [robot_state_publisher]: got segment ee_link
[robot_state_publisher-1] [INFO] [1620150401.147125000] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-1] [INFO] [1620150401.147141100] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-1] [INFO] [1620150401.147152000] [robot_state_publisher]: got segment tool0
[robot_state_publisher-1] [INFO] [1620150401.147161800] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-1] [INFO] [1620150401.147172600] [robot_state_publisher]: got segment world
[robot_state_publisher-1] [INFO] [1620150401.147182600] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-1] [INFO] [1620150401.147192400] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-1] [INFO] [1620150401.147201000] [robot_state_publisher]: got segment wrist_3_link
```

### terminal 3:
$ `ros2 run gazebo_ros spawn_entity.py -topic /robot_description -entity ur10`

```
[INFO] [1620150404.420155200] [spawn_entity]: Spawn Entity started
[INFO] [1620150404.420965700] [spawn_entity]: Loading entity published on topic /robot_description
[INFO] [1620150404.424194700] [spawn_entity]: Waiting for entity xml on /robot_description
[INFO] [1620150404.686556400] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[INFO] [1620150404.687121000] [spawn_entity]: Waiting for service /spawn_entity
[INFO] [1620150404.698377300] [spawn_entity]: Calling service /spawn_entity
[INFO] [1620150404.995159600] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [ur10]
```

Are any steps missing?

FelipeGdM, 2021-05-07:
Hi @newcanopies!

Could you print the value of `GAZEBO_MODEL_PATH` and `GAZEBO_RESOURCE_PATH` environment variables?

They store the list of directories where Gazebo will search for files to build your model (see the [guide](http://gazebosim.org/tutorials?tut=components&cat=get_started#EnvironmentVariables) for more). 

> [Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ur10_gazebo/models/ur10/meshes/dae/wrist3.dae]
[Err] [MeshShape.cc:64] Failed to find mesh file [model://ur10_gazebo/models/ur10/meshes/dae/wrist3.dae]
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_ros_control.so: 
libgazebo_ros_control.so: cannot open shared object file: No such file or directory
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ur10_gazebo/models/ur10/meshes/dae/upperarm.dae]

My initial guess is that this URI `model://ur10_gazebo/models/ur10/meshes/dae/wrist3.dae` is bad formatted or `GAZEBO_MODEL_PATH` was not updated accordingly. Usually, there is no `models/` folder within a model folder, it seems that a `package://` was translated to `model://` (maybe during urdf > sdf convertion?). Strictly speaking, it is not wrong, but it defines how `GAZEBO_MODEL_PATH` should be updated

> [Err] [Visual.cc:2956] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [$(find ur10)/models/ur10/meshes/stl/upperarm.stl]

This is strange, because Gazebo should not "see" the `$(find ur10)` directive, since it is a `.xacro` command. Have you updated the `.xacro` file after it was parsed?

newcanopies, 2021-05-07:
Ola @FelipeGdM 

Thanks very much for looking into this. There are other two people reporting the same exact issue that when spawning a URDF, the meshes are not visualized. However, I am afraid that it is a non trivial bug, not just a simple omission in steps taken, or xml syntax. 

Since everything points to a bug or a missing-capability of the spawn_entity tool, I placed the minimal demo in this repo: https://github.com/newcanopies/ur10_spawn_demo

It consists of a launch file, and the URDF model, and the meshes have absolute `package://` paths. 

>  it seems that a `package://` was translated to `model://`

Yes, this is the peculiar thing. The mesh URI's are absolute using `<mesh filename="package://...`  but Gazebo logs them as  `model://`

The reason I also tried pointing to the meshes with the `$(find ur10)` directive, is that it allows the spawning to complete much faster than using `package://` directive.

echo $GAZEBO_RESOURCE_PATH
/usr/share/gazebo-11:
echo $GAZEBO_MODEL_PATH
/ws_dev/ur10_spawn_demo/models/: and/other/exported/paths: ....

Do you think that it's a bug with spawn_entity.py? 
Can you think of a mesh based URDF that you have seen spawned in Gazebo11?

brennand, 2021-06-02:
I'm also having this problem.  My URDF works fine in ros1 and if I replace the mesh with an box it works fine.  Also the URDF is displayed properly in RVIZ.  But when I run the command gazebo freezes.:

`
ros2 run gazebo_ros spawn_entity.py -topic /robot_description -entity robot

[INFO] [1622639718.137983415] [spawn_entity]: Spawn Entity started

[INFO] [1622639718.138268170] [spawn_entity]: Loading entity published on topic /robot_description

[INFO] [1622639718.139824302] [spawn_entity]: Waiting for entity xml on /robot_description

[INFO] [1622639718.250162743] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30

[INFO] [1622639718.251562194] [spawn_entity]: Waiting for service /spawn_entity

[INFO] [1622639718.266057352] [spawn_entity]: Calling service /spawn_entity

[INFO] [1622639718.464459423] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [robot]
`

Using foxy + gazebo 11.5.1  all via apt

brennand, 2021-06-07:
> I'm also having this problem, where meshes are making gazebo to crash. My URDF works fine in ros1 and if I replace the mesh with an box it works fine. I've also tried with stl and .dae files and they both make gazebo to crash. The URDF is displayed properly in RVIZ. But when I run the command gazebo freezes.:
> 
> `$ ros2 run gazebo_ros spawn_entity.py -topic /robot_description -entity robot`
> 
> ```
> [INFO] [1622639718.137983415] [spawn_entity]: Spawn Entity started
> [INFO] [1622639718.138268170] [spawn_entity]: Loading entity published on topic /robot_description
> [INFO] [1622639718.139824302] [spawn_entity]: Waiting for entity xml on /robot_description
> [INFO] [1622639718.250162743] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
> [INFO] [1622639718.251562194] [spawn_entity]: Waiting for service /spawn_entity
> [INFO] [1622639718.266057352] [spawn_entity]: Calling service /spawn_entity
> [INFO] [1622639718.464459423] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [robot]
> ```
> 
> urdf snipet
> 
> ```
>   <link name="joint">
>     <visual>
>         <origin xyz="0 0 0" rpy="0 0 0"/>
>         <geometry>
>             <mesh filename="package://robot_description/meshes/joint/joint.dae"/>
>         </geometry>
>         <material name="cabinet_color"/>
>     </visual>
>     <collision>
>       <origin xyz="0 0 0" rpy="0 0 0"/>
>       <geometry>
>           <mesh filename="package://robot_description/meshes/joint/joint.dae"/>
>       </geometry>
>     </collision>
>     <inertial>
>       <mass value="1"/>
>       <inertia  ixx="0.001" ixy="0.0"  ixz="0.0"
>                 iyy="0.001"  iyz="0.0"
>                 izz="0.001"/>
>     </inertial>
>   </link>
> ```
> 
> Using foxy + gazebo 11.5.1 all via apt

So I got it working in the end.  Was coursed by wrong gazebo paths.  But would have been nice to have a warning.

newcanopies, 2021-06-07:
> The package:// was getting converted to a model:// So I had to copy my package to the model path.

Hi @brennand, please describe your steps:

* where did you copy your package `robot_description/robot/meshes` to?
* did you convert the path to URDF meshes from package:// to model:// ?
* did you 
export  GAZEBO_MODEL_PATH=~/ROS2_ws/robot_description/robot/models/meshes:${GAZEBO_MODEL_PATH}  ? 
* any other changes that made Gazebo 11 load meshes from spawned URDF?

newcanopies, 2021-06-07:
@brennand @j-rivero the suggested change of the URDF mesh URI from package:// to model://  does not work 
(tested with https://github.com/newcanopies/ur10_spawn_demo)

newcanopies, 2021-06-07:
> The URDF is displayed properly in RVIZ. But when I run the command gazebo freezes.:
> 
> `$ ros2 run gazebo_ros spawn_entity.py -topic /robot_description -entity robot`

hi @brennand 
does your URDF spawning method require RViz to also be running?
or you have another script publishing the -entity robot on -topic /robot_description?

brennand, 2021-06-09:
> hi @brennand
> does your URDF spawning method require RViz to also be running?
> or you have another script publishing the -entity robot on -topic /robot_description?

I just added the ROS package that has the meshes to this folder. `~/.gazebo/models`  This is the default folder for gazebo models.  So didn't need need to change any $PATHS eta.  

On a side note for people coming here.  I also added some code to the launch.py that made the symbolic link to the models folder.  So other people on the team didn't need to add them by hand.

newcanopies, 2021-06-09:
thanks very much @brennand 

two last bits, 
*1
does that mean you did not change anything in the URDF pointer to the meshes, 
does your URDF mesh filename still point to the original location  and **not** to `~/.gazebo/models` 

```
<geometry>
            <mesh filename="package://robot_description/meshes/joint/joint.dae"/>
 </geometry>
```

2* 
Is your Gazebo installation source packages.**osrfoundation**.org or packages.ros.org?
please check with, apparently the install source affects functionality
$ `apt-cache   policy gazebo11`

newcanopies, 2021-06-09:
hi @j-rivero

this URDF mesh URI ([rosanswers](https://answers.ros.org/question/373495/ros2-foxy-gazebo-spawn_entity-systempathscc459-file-or-path-does-not-exist/?answer=373549#post-id-373549))
```
<geometry>
            <mesh filename="file://$(find ur10_spawn_demo)/models/ur10_urdf/meshes/visual/base.dae"/>
 </geometry>
```

throws this error:
```
[gazebo-1] [Err] [MeshShape.cc:64] Failed to find mesh file [file://$(find ur10_spawn_demo)/models/ur10_urdf/meshes/visual/base.dae]
```

MeshShape.cc:64 https://github.com/osrf/gazebo/blob/5e2f41462108657694165d63b00e55022ec95007/gazebo/physics/MeshShape.cc#L49-L70
```
  auto uri = common::asFullPath(this->sdf->Get<std::string>("uri"),
      this->sdf->FilePath());

  this->sdf->GetElement("uri")->Set(uri);

  std::string meshStr = uri;
  common::MeshManager *meshManager = common::MeshManager::Instance();
  this->mesh = meshManager->GetMesh(meshStr);

  if (!this->mesh)
  {
    meshStr = common::find_file(uri);

    if (meshStr == "__default__" || meshStr.empty())
    {
      gzerr << "Failed to find mesh file [" << uri << "]" << std::endl;
      return;
    }

    if ((this->mesh = meshManager->Load(meshStr)) == NULL)
      gzerr << "Unable to load mesh from file[" << meshStr << "]\n";
  }
```

If I make a simple `mesh-sphere.STL` mesh , what is the correct URDF mesh URI that can spawn it in Gazebo?

closed 2021-06-09 by newcanopies

bjsowa, 2021-06-09:
Why was this issue closed?

newcanopies, 2021-06-09:
This issue is a **BUG** involving [Visual.cc:2956](https://github.com/osrf/gazebo/blob/5e2f41462108657694165d63b00e55022ec95007/gazebo/rendering/Visual.cc#L2955-L2956), [SystemPaths.cc:459](https://github.com/osrf/gazebo/blob/5e2f41462108657694165d63b00e55022ec95007/gazebo/common/SystemPaths.cc#L457-L462), [MeshShape.cc:64](https://github.com/osrf/gazebo/blob/5e2f41462108657694165d63b00e55022ec95007/gazebo/physics/MeshShape.cc#L49-L70)

but the above https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1272#issuecomment-857513366 quick-fix works

bjsowa, 2021-06-09:
> This issue is a **BUG** involving [Visual.cc:2956](https://github.com/osrf/gazebo/blob/5e2f41462108657694165d63b00e55022ec95007/gazebo/rendering/Visual.cc#L2955-L2956), [...]

How do you know that? Why is the `spawn_model` script for ROS working with `package://` references but `spawn_entity.py` from ROS2 port does not work, even though both distributions use the same version of Gazebo?

newcanopies, 2021-06-10:
To the best of my knowledge, having tested every possible method spawning with `gazebo_ros` via launch scripts and command line using ROS1 validated UR10 URDF meshes and every URI syntax (file: $find, model:, package:) none of which are the solution. 

The spawning method changed in ROS2, and the three files linked above indicate that Gazebo11 is not aware of ROS2 system paths, which is why exporting the GAZEBO_MODEL_PATH does not find the meshes in the workspace, but only in the default Gazebo models directory. The `sdformat.log` steps through the parsing and conversion of the spawned URDF to SDF under the Gazebo hood, have a look there for your use case. 

Of course this Issue should be addressed in a PR to avoid maintaining **all copies** of robot_descriptions under Gazebo database.
@bjsowa @brennand @j-rivero

andrestoga, 2022-11-27:
> > > The URDF is displayed properly in RVIZ. But when I run the command gazebo freezes.:
> > > `$ ros2 run gazebo_ros spawn_entity.py -topic /robot_description -entity robot`
> > 
> > 
> > hi @brennand
> > does your URDF spawning method require RViz to also be running?
> > or you have another script publishing the -entity robot on -topic /robot_description?
> 
> I just added the ROS package that has the meshes to this folder. `~/.gazebo/models` This is the default folder for gazebo models. So didn't need need to change any $PATHS eta.
> 
> On a side note for people coming here. I also added some code to the launch.py that made the symbolic link to the models folder. So other people on the team didn't need to add them by hand.
> 
> ```
>     spawn_robot = Node(
>         package="gazebo_ros",
>         executable="spawn_entity.py",
>         arguments=["-topic", "robot_description", "-entity", "robot", "-x", "0.0", "-y", "0.0", "-z", "0.0"],
>         output="screen")
> ```
> 
> This is how I "Spawn the robot"
> 
> P.S in the ~/.gazebo folder you should find the logs with some error msgs.

How did you add the symbolic link in the launch.py to the models folder?

andrestoga, 2022-12-22:
> To the best of my knowledge, having tested every possible method spawning with `gazebo_ros` via launch scripts and command line using ROS1 validated UR10 URDF meshes and every URI syntax (file: [...]

Where is the `sdformat.log`?

aihoque2, 2023-02-21:
I learned that I need to transfer my package to `~/.gazebo/models`, but how do I tell my launch file to do that?

Pavankv92, 2023-10-31:
> "-x", "0.0", "-y", "0.0", "-z", "0.0"

Add this line to the launch file

XML launch file,
 <set_env name="GAZEBO_MODEL_PATH" value="$(find-pkg-prefix  robot_description)/share/"  />

Python launch file,

from launch.actions import SetEnvironmentVariable
gazebo_env = SetEnvironmentVariable("GAZEBO_MODEL_PATH", os.path.join(
    #     get_package_prefix("robot_description"), "share"))

With above line, meshes can still used with "package://"