ros-simulation/gazebo_ros_pkgs #1279: Unable to build gazebo_ros_pkgs with ros2 eloquent on ubuntu 18.04

arpit6232, 2021-05-28:
gazebo_ros fails to colcon build under gtest error.

**Description**

Steps followed to build gazebo_ros_pkgs for ros2 eloquent 
In the dev_ws which sources 
1. wget https://raw.githubusercontent.com/ros-simulation/gazebo_ros_pkgs/ros2/gazebo_ros_pkgs.repos
2. vcs import src < gazebo_ros_pkgs.repos
3. vcs custom --args checkout eloquent
4. rosdep install --from-paths src --ignore-src -r -y
5. 

Fails with the following output on terminal :
[error.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/6557740/error.txt)

**To Reproduce**

1. wget https://raw.githubusercontent.com/ros-simulation/gazebo_ros_pkgs/ros2/gazebo_ros_pkgs.repos
2. vcs import src < gazebo_ros_pkgs.repos
3. vcs custom --args checkout eloquent
4. rosdep install --from-paths src --ignore-src -r -y

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
To perfectly build gazebo_ros_msgs

A clear and concise description of what you expected to happen.

**Screenshots**
```
Starting >>> gazebo_ros
--- stderr: gazebo_ros                               
In file included from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67:0,
                 from /home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15:
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:38: warning: ‘constexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [-Wdeprecated-declarations]
   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
                                      ^
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:38: note: in definition of macro ‘INSTANTIATE_TEST_CASE_P’
   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:62:0,
                 from /home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15:
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16: note: declared here
 constexpr bool InstantiateTestCase_P_IsDeprecated() { return true; }
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67:0,
                 from /home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15:
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:496:73: warning: ‘constexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [-Wdeprecated-declarations]
   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
                                                                         ^
/home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1: note: in expansion of macro ‘INSTANTIATE_TEST_CASE_P’
 INSTANTIATE_TEST_CASE_P(Plugins, TestPlugins, ::testing::Values(
 ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:62:0,
                 from /home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15:
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16: note: declared here
 constexpr bool InstantiateTestCase_P_IsDeprecated() { return true; }
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67:0,
                 from /home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15:
/home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp: In function ‘std::__cxx11::string gtest_PluginsTestPlugins_EvalGenerateName_(const testing::TestParamInfo<TestParams>&)’:
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:472:43: error: expected primary-expression before ‘)’ token
       auto t = std::make_tuple(__VA_ARGS__);                                  \
                                           ^
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:498:3: note: in expansion of macro ‘INSTANTIATE_TEST_SUITE_P’
   INSTANTIATE_TEST_SUITE_P
   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1: note: in expansion of macro ‘INSTANTIATE_TEST_CASE_P’
 INSTANTIATE_TEST_CASE_P(Plugins, TestPlugins, ::testing::Values(
 ^~~~~~~~~~~~~~~~~~~~~~~
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:473:48: error: template argument 1 is invalid
       static_assert(std::tuple_size<decltype(t)>::value <= 2,                 \
                                                ^
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:498:3: note: in expansion of macro ‘INSTANTIATE_TEST_SUITE_P’
   INSTANTIATE_TEST_SUITE_P
   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1: note: in expansion of macro ‘INSTANTIATE_TEST_CASE_P’
 INSTANTIATE_TEST_CASE_P(Plugins, TestPlugins, ::testing::Values(
 ^~~~~~~~~~~~~~~~~~~~~~~
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:479:23: error: expected primary-expression before ‘)’ token
         DUMMY_PARAM_))))(info);                                               \
                       ^
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:498:3: note: in expansion of macro ‘INSTANTIATE_TEST_SUITE_P’
   INSTANTIATE_TEST_SUITE_P
   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1: note: in expansion of macro ‘INSTANTIATE_TEST_CASE_P’
 INSTANTIATE_TEST_CASE_P(Plugins, TestPlugins, ::testing::Values(
 ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest-param-test.h:181:0,
                 from /home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:67,
                 from /home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:15:
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-param-util.h: In instantiation of ‘void testing::internal::TestNotEmpty() [with T = int]’:
/home/arps/dev_ws/src/gazebo_ros_pkgs/gazebo_ros/test/test_plugins.cpp:70:1:   required from here
/home/arps/ros2_eloquent/install/gtest_vendor/src/gtest_vendor/include/gtest/internal/gtest-param-util.h:382:3: error: static assertion failed: Empty arguments are not allowed.
   static_assert(sizeof(T) == 0, "Empty arguments are not allowed.");
   ^~~~~~~~~~~~~
make[2]: *** [test/CMakeFiles/test_plugins.dir/test_plugins.cpp.o] Error 1
make[1]: *** [test/CMakeFiles/test_plugins.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< gazebo_ros [3.67s, exited with code 2]
```
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

 - OS: Ubuntu 18.04
 - Gazebo version:  Gazebo 9
 - ROS version: ROS2 eloquent
  - gazebo_ros_pkgs: source install
 - glx server in use: output from `glxinfo | grep 'server glx vendor'` (glxinfo is in mesa-utils package for Debian/Ubuntu)

**Additional context**

Add any other context here.

arpit6232, 2021-05-30:
This issue is only present when trying to install from source, and does not arise if apt package manager is used to install gazebo_pkgs.

closed 2021-06-09 by arpit6232

scpeters, 2021-07-13:
I recently experienced this problem and I believe it was related to sourcing the wrong setup file (foxy, when I meant to only use galactic)