ros-simulation/gazebo_ros_pkgs #1287: [ROS2] Unable to get object pose with `/get_entity_state` service

nicholaspalomo, 2021-06-16:
**Description**

ROS2 hangs upon calling `/get_entity_state` message to query pose information about model spawned in Gazebo.

**To Reproduce**

Steps to reproduce the behavior:
1. `source /opt/ros/foxy/setup.bash`
2. `ros2 launch gazebo_ros gazebo.launch.py`
3. (If your Gazebo window doesn't open automatically, open it with `gzclient`)
4. Run the following script:
```
import sys

import rclpy
from rclpy.node import Node
from rclpy.client import Client

from gazebo_msgs.srv import SpawnEntity, GetEntityState
from geometry_msgs.msg import Pose

MODEL_DATABASE_TEMPLATE = """\
<sdf version="1.4">
    <world name="default">
        <include>
            <uri>model://{}</uri>
        </include>
    </world>
</sdf>"""

def spin_until_service_complete(node, response):
    rclpy.spin_until_future_complete(node, response)
    if response.result() is not None:
        node.get_logger().info('SERVICE COMPLETE! RESULT:\n{}'.format(response.result()))
        return response.result()
def main():
    rclpy.init(args=sys.argv)
    node: Node = rclpy.create_node('sandbox')

    spawn_client: Client = node.create_client(SpawnEntity, '/spawn_entity')

    spawn_request: SpawnEntity.Request = SpawnEntity.Request()
    spawn_request.name = "cube"
    spawn_request.xml = MODEL_DATABASE_TEMPLATE.format('wood_cube_5cm')
    spawn_request.reference_frame = "world"
    spawn_request.initial_pose = Pose()
    spawn_request.initial_pose.position.x = 0.4 # [m]
    spawn_request.initial_pose.position.y = 0.0 # [m]
    spawn_request.initial_pose.position.z = 0.025 # [m]

    node.get_logger().info('TRYING TO SPAWN CUBE INTO GAZEBO!')
    response = spawn_client.call_async(spawn_request)
    spin_until_service_complete(node, response)

    node.get_logger().info('CUBE SUCCESSFULLY SPAWNED!')

    get_entity_state_client: Client = node.create_client(GetEntityState, '/get_entity_state')
    get_entity_state_request: GetEntityState.Request = GetEntityState.Request()
    get_entity_state_request.name = "cube"
    get_entity_state_request.reference_frame = "world"

    node.get_logger().info('TRYING TO GET CUBE POSE!')
    response: GetEntityState.Response = get_entity_state_client.call_async(get_entity_state_request)

    result = spin_until_service_complete(node, response)
    node.get_logger().info('GOT CUBE POSE!\n{}'.format(result))

    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

**Expected behavior**

`/get_entity_state` should return a `GetEntityState.Response` containing the pose information about the cube (or whatever object) you spawn in Gazebo.

**Environment:**

 - OS: Ubuntu 20.04
 - Gazebo version: Gazebo 11.6.0
 - ROS version: Foxy
 - gazebo_ros_pkgs: debian install
 - glx server in use: NVIDIA

solid-sinusoid, 2022-01-18:
@nicholaspalomo Hello, did you solve this problem? Now I myself need to implement this service. And I would like to know your feedback

swuuu, 2022-05-17:
Hi, I was also trying to read the robot pose from gazebo and could not find the /get_entity_state service. I found a workaround, which is to add the `lib_gazebo_ros_state.so` plugin in the world file being launched. (In my case, it was /usr/share/gazebo-11/worlds/empty.world). This link helped me: [https://answers.ros.org/question/360161/ros2-dashing-service-get_entity_state-is-missing/](url).