ros-simulation/gazebo_ros_pkgs #1293: How to install "gazebo_ros_pkgs" on MacOS

qixin1106, 2021-07-06:
MacOS 11.4 (20F71)

```
qixin@LocalQixiniMac gazebo_ros_pkgs-3.5.3 % colcon build
Starting >>> gazebo_dev
Starting >>> gazebo_msgs
Finished <<< gazebo_dev [0.39s]                                  
Finished <<< gazebo_msgs [6.58s]                     
Starting >>> gazebo_ros
--- stderr: gazebo_ros                               
In file included from /Users/qixin/Documents/ROS/gazebo_ros_pkgs-3.5.3/src/gazebo_ros/src/gazebo_ros_init.cpp:21:
In file included from /usr/local/include/gazebo-11/gazebo/transport/Node.hh:30:
In file included from /usr/local/include/gazebo-11/gazebo/transport/TopicManager.hh:35:
In file included from /usr/local/include/gazebo-11/gazebo/transport/SubscriptionTransport.hh:24:
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:57:65: error: expected class name
    class GZ_TRANSPORT_VISIBLE ConnectionReadTask : public tbb::task
                                                                ^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:73:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:313:61: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                  ConnectionReadTask *task = new(tbb::task::allocate_root())
                                                 ~~~~~~~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:315:19: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                  tbb::task::enqueue(*task);
                  ^~~~~~~~~~~~~~~~~~
                  ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
In file included from /Users/qixin/Documents/ROS/gazebo_ros_pkgs-3.5.3/src/gazebo_ros/test/test_gazebo_ros_state.cpp:15:
In file included from /usr/local/include/gazebo-11/gazebo/test/ServerFixture.hh:44:
In file included from /usr/local/include/gazebo-11/gazebo/transport/transport.hh:3:
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:57:65: error: expected class name
    class GZ_TRANSPORT_VISIBLE ConnectionReadTask : public tbb::task
                                                                ^
In file included from /Users/qixin/Documents/ROS/gazebo_ros_pkgs-3.5.3/src/gazebo_ros/src/gazebo_ros_init.cpp:21:
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:39:58: error: expected class name
    class GZ_TRANSPORT_VISIBLE PublishTask : public tbb::task
                                                         ^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:73:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:54:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:313:61: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                  ConnectionReadTask *task = new(tbb::task::allocate_root())
                                                 ~~~~~~~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:162:52: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                PublishTask *task = new(tbb::task::allocate_root())
                                        ~~~~~~~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:315:19: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                  tbb::task::enqueue(*task);
                  ^~~~~~~~~~~~~~~~~~
                  ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:165:17: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                tbb::task::enqueue(*task);
                ^~~~~~~~~~~~~~~~~~
                ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
In file included from /Users/qixin/Documents/ROS/gazebo_ros_pkgs-3.5.3/src/gazebo_ros/test/test_gazebo_ros_state.cpp:15:
In file included from /usr/local/include/gazebo-11/gazebo/test/ServerFixture.hh:44:
In file included from /usr/local/include/gazebo-11/gazebo/transport/transport.hh:6:
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:39:58: error: expected class name
[... 168 lines cut ...]
make[1]: *** [CMakeFiles/gazebo_ros_init.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/gazebo_ros_factory.dir/src/gazebo_ros_factory.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_ros_factory.dir/all] Error 2
8 errors generated.
make[2]: *** [CMakeFiles/gazebo_ros_properties.dir/src/gazebo_ros_properties.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_ros_properties.dir/all] Error 2
8 errors generated.
make[2]: *** [test/CMakeFiles/test_gazebo_ros_link_wrench.dir/test_gazebo_ros_link_wrench.cpp.o] Error 1
make[1]: *** [test/CMakeFiles/test_gazebo_ros_link_wrench.dir/all] Error 2
8 errors generated.
make[2]: *** [test/CMakeFiles/test_gazebo_ros_properties.dir/test_gazebo_ros_properties.cpp.o] Error 1
make[1]: *** [test/CMakeFiles/test_gazebo_ros_properties.dir/all] Error 2
8 errors generated.
make[2]: *** [test/CMakeFiles/test_gazebo_ros_state.dir/test_gazebo_ros_state.cpp.o] Error 1
make[1]: *** [test/CMakeFiles/test_gazebo_ros_state.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< gazebo_ros [12.7s, exited with code 2]

Summary: 2 packages finished [19.7s]
  1 package failed: gazebo_ros
  1 package had stderr output: gazebo_ros
  2 packages not processed
```

j-rivero, 2021-07-06:
There were some problems with tbb versions in Brew and other systems. The great @scpeters patched Gazebo in https://github.com/osrf/gazebo/pull/3037 , not sure if we need a new release.

jasonhan-vassar, 2021-08-13:
I'm also running into the same issue using gazebo11 and ROS 2 Foxy. I'm on macOS 11.1.

scpeters, 2021-08-13:
for now we have to use an old version of `tbb` in our home-brew fork, which requires setting some extra environment variables, since it does not install to `/usr/local`. see the changes in https://github.com/ignition-tooling/release-tools/pull/466 of the following lines:

https://github.com/ignition-tooling/release-tools/blob/53b72f3c2fcb7d01613f028014e915760f59b96c/jenkins-scripts/lib/project-default-devel-homebrew-amd64.bash#L119-L121