ros-simulation/gazebo_ros_pkgs #1301: [ros2] gazebo plugins should publish /tf topic as /namespace/tf if the robot is spawned with namespace

sathak93, 2021-08-01:
gazebo_plugins publishes  /odom, /scan, /imu and other topics with namespaces but not the  /tf topic when spawned the robot with spawn_entity.py
[image]

bjsowa, 2021-08-02:
The `/tf` and `/tf_static` are always published on the global namespace.

sathak93, 2021-08-03:
@bjsowa  how to configure global namespace in launch file.
also  robot_state_publisher also uses tf framework why would it publishes with namespace if it is started with namespace?.
 if we spawn multiple robots how do we differentiate the /tf topipc from one another.

SteveMacenski, 2021-08-03:
Also this is useful for separations for multi-robot configurations. The spawner we have in Nav2 creates the TF remap. Ideally, we'd like to delete that and use the gazebo_ros's but the blocking reason we can't is that we cannot remap the tf.

bjsowa, 2021-08-06:
> @bjsowa how to configure global namespace in launch file.
> also robot_state_publisher also uses tf framework why would it publishes with namespace if it is started with namespace?.
> if we spawn multiple robots how do we differentiate the /tf topipc from one another.
> [image]

I don't know how you launched the `robot_state_publisher` but when I tried running it under a namespace, the `/tf` topic stays under global namespace:

[image]

bjsowa, 2021-08-06:
If you look at the code of tf2_ros::TransformBroadcaster:
https://github.com/ros2/geometry2/blob/5f30fe8fcc9eacde580fb1374ae82182e7fb5bb6/tf2_ros/include/tf2_ros/transform_broadcaster.h#L58-L61

you will notice it creates a publisher on the `/tf` topic. The leading slash makes the topic be resolved in the global namespace (it won't be pushed to the node's namespace). The same goes for TransformListener. 

This is the intended behavior. Here are parts of the documentation that I could find that explain why it was designed that way:

> The concept of tf `frame_ids` is not scoped in the same way as ROS Names. In particular, namespacing a specific subpart of a computation graph does not change the physical layout which the tf tree represents. Because of this `frame_ids` do not follow namespace remapping rules.

> Why don't frame_ids use ROS namespaces?
> There is already a hierarchy defined by the tf graph in the child/parent relationships, vs. ROS names where the hierarchy is defined by the name.
>
> namespaces are used to determine how communication channels are connected. Whereas tf frame_ids are related to how coordinate frames are physically connected. Thus if you push down a set of nodes relating to the 2d navigation they will work in their own namespace and not have collisions on topics or services. However, the base is still connected to the body in the same way, despite the namespace push down. Thus pushing down doesn't make sense.
>
> There are a few other considerations, tf frames are not necessarily connected to any specific node. For example on the pr2 the base_laser frame is published by the robot_state_publisher based on the URDF, but the hokuyo_node publishes scans in the base_laser frame. And then rviz transforms the data into some arbitrary frame for viewing. If any of them disagree about the full name of "base_laser" this will not work, for the tree in tf will not be connected anymore.

SteveMacenski, 2021-08-06:
I would typically agree with you, but for our needs we need to have multiple robots running around (hence the spawner) and in that situation having all of the frames on the same TF topic / buffer breaks the separation of the robot systems that we also wouldn't want to be true on hardware. 

We remap the `/tf` to `tf` to accomplish that and then have `tf` get thrown into the appropriate namespace.