ros-simulation/gazebo_ros_pkgs #1305: Ros2: Set Gazebo clock publish rate

HarunTeper, 2021-08-20:
Hello,

the default publish rate of the gazebo clock node is 10Hz. So far, it is most convenient to use the following approach to use Ros2 and Gazebo in a launch file for me:

gazebo_launch_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(pkg_gazebo_ros, 'launch', 'gazebo.launch.py')),
        launch_arguments={'world': world}.items()
    )
However, looking at the launch files and some other threads, I can't manage to set the publish_rate for the gazebo clock node. Is it possible to include the parameter .yaml file in the launch arguments or to include a parameter 'publish_rate': '10'} to the list of parameters, so that it can be set by the launch file.

allenh1, 2021-11-11:
This is a duplicate of #1211 and #1247... Would be really great if this could be addressed. It makes the gazebo clock unusable for time-sensitive calculations.

john-maidbot, 2021-12-07:
I had this same issue. Pretty frustrating :sweat_smile: If anyone else finds this, there is a workaround (not really a solution, but it worked for me on ros2 galactic, gazebo version 11.9.0): https://answers.ros.org/question/378362/how-to-set-gazebo-clock-publish-rate-foxy/

allenh1, 2021-12-11:
@john-maidbot I have a pull request (see #1332) for dealing with this... It works for all my purposes, but would require you to build from source.

jacobperron, 2022-05-17:
https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1391 has made it more convenient to set ROS parameters (including `publish_rate`) from the launch file.

closed 2022-05-17 by jacobperron

pinkponk, 2023-10-03:
For anyone else having problems with this and sees this issue, I finally got it to work with this:

```python
    gazebo_params_path = os.path.join(
                    get_package_share_directory(package_name),'config','gazebo_params.yaml')
    gazebo_world_path = os.path.join(
                    get_package_share_directory(package_name),'worlds','AAAA.world')

    gazebo = IncludeLaunchDescription(
                PythonLaunchDescriptionSource([os.path.join(
                    get_package_share_directory('gazebo_ros'), 'launch', 'gazebo.launch.py')]),
                    launch_arguments={
                        'params_file': gazebo_params_path,
                                      'world': gazebo_world_path }.items()
         )
```

Dont forget to include the config dir in the CMakeList.txt
```
install(
  DIRECTORY config launch worlds
  DESTINATION share/${PROJECT_NAME}
)
```

gazebo_config.yaml
```yaml
# gazebo_params.yml
gazebo:
    ros__parameters:
        publish_rate: 400.0
```

PR #1391 (2022-05-17): Expose launch parameter for `params_file`
This PR exposes a `params_file` launch argument for `gazebo` with which the clock rate (or any other ROS parameters) can be set.

For example:

``` yaml
# params.yml
gazebo:
    ros__parameters:
        publish_rate: 400.0
```

`ros2 launch gazebo_ros gazebo.launch.py params_file:=params.yml`

See [wiki](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration%3A-ROS-Clocks-and-sim-time) for more

- Closes #1305 
- Closes #1211 
- Closes #1247