ros-simulation/gazebo_ros_pkgs #1310: Average rate of transform between the odom and the base link is slow.

padhupradheep, 2021-09-16:
Hey, 

I'm using the differential drive plugin to simulate a robot in ROS 2. The distro used is Rolling. 

I wanted transform between the odom and the base_footprint of the robot, so I use the lookup transform from tf2. I found that the transformation between the "odom" and the "base_link" is being published very slow. Here is the output of the tf2_monitor:

`[INFO] [1631802573.358697219] [tf2_monitor_main]: Waiting for transform odom ->  base_footprint: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
Gathering data on odom -> base_footprint for 10 seconds...

RESULTS: for odom to base_footprint
Chain is: odom -> base_footprint
Net delay     avg = 3.10486e+08: max = 1.6318e+09

Frames:
Frame: base_footprint, published by <no authority available>, Average Delay: 1.6318e+09, Max Delay: 1.6318e+09
`

What could be the reason for it?

omerts, 2022-06-28:
Adding to the above:
[image]
You can see the average rate is 10. This also causes EKF filter node to not have quick enough data.

KenStonck, 2023-12-14:
I am facing the same issues with my setup. Have you been able to find a solution ?