ros-simulation/gazebo_ros_pkgs #1321: <tf2_geometry_msgs/tf2_geometry_msgs.h> not found

ItsJas, 2021-10-12:
Hi,

So I was trying to install gazebo from source today on my Ubuntu 20 running ROS Galactic following [these steps](http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros) using the ros2 branch. And I got a `"<tf2_geometry_msgs/tf2_geometry_msgs.hpp> not found error"`.

I was pretty sure that I have the tf2_geometry_msgs package installed already so when I tried looking for the missing header file in my system, it was actually `tf2_geometry_msgs.h`. So the build only worked by replacing  `tf2_geometry_msgs.hpp` with `tf2_geometry_msgs.h` everywhere needed.

Not sure if that's the proper fix though so I wanted to report here!

Thanks,
Jas

padhupradheep, 2021-11-02:
Hey, ros2 branch is intended for Rolling distribution I believe. You need to clone the galactic branch, where the headers are in place as you have described!

ItsJas, 2021-11-04:
@padhupradheep Sorry I didn't notice the Galactic branch. Will use that then.
Thank you!

closed 2021-11-04 by ItsJas