ros-simulation/gazebo_ros_pkgs #1327: [URGENT] Gazebo not launching on ros2- rolling distribution

poornimajd, 2021-11-04:
I tried using gazebo in ros2 rolling distribution in a docker container(ubuntu 20.04)But whenever I try to launch the nav2 stack using the command- 
`ros2 launch nav2_bringup tb3_simulation_launch.py`
I get the following error- 
`gzserver: symbol lookup error: /opt/ros/rolling/lib/libgazebo_ros_init.so: undefined symbol: _ZN6rclcpp11InitOptionsC1E19rcutils_allocator_t`

The gazebo version is 11.5.1.
I tried to check out sources online,but could not resolve this error.

Any suggestion is appreciated.
Thank you

chapulina, 2021-11-04:
How did you install nav2? It's not on rolling yet (https://github.com/ros-planning/navigation2/issues/2648). It would be helpful if you shared your docker container.

poornimajd, 2021-11-05:
I resolved the issue,by installing sudo apt-get install ros-rolling-gazebo* .Even though gazebo_ros_pkgs was installed for the rolling distribution .Some packages were missing ,hence the error.
Thank you so much for the speedy reply.

closed 2021-11-05 by poornimajd