ros-simulation/gazebo_ros_pkgs #1329: Use x-forward z-up frame for pointcloud

ViniciusAbrao, 2021-11-05:
The issue fixed for the melodic-devel found here:

https://github.com/sachazyto/gazebo_ros_pkgs/commit/1a038579dd256505a1786a4e8566dab25ec2f48f

It would be also necessary to change for the foxy and galactic. I assume it should be done at lines 747 and 748 in the following:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/galactic/gazebo_plugins/src/gazebo_ros_camera.cpp

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/foxy/gazebo_plugins/src/gazebo_ros_camera.cpp

Otherwise the pointcloud2 really has shown rotated in the rviz.

ViniciusAbrao, 2021-11-09:
I have implemented and tested the solution mentioned above here:

https://github.com/ViniciusAbrao/ros2_depth_cam_plugin

It really works now as expected. Maybe it would be a good PR. I am going to open it.

peci1, 2023-06-14:
This is not a good way of solving this issue.

REP-103 says:

> In the case of cameras, there is often a second frame defined with a "_optical" suffix. This uses a slightly different convention:

>    z forward
>    x right
>    y down

So, basically, each camera should have its base geometrical frame (`camera`) and an accompanying optical frame (`camera_optical`), which is rotated by `rpy="-pi/2 0 -pi/2"` relative to the base frame. The depth cloud should have the optical frame in its header.