ros-simulation/gazebo_ros_pkgs #1345: ROS 2 Ackermann steering. Angle vs angular velocity

joe28965, 2022-01-11:
I noticed something when working with the ackermann plugin for ROS 2 (Galactic). It seems to work fine when driving by hand, it seems to follow the commands that I give it using teleop. However, as soon as I use something like Nav2 it cannot follow the planned path. So I did some digging.

The line that calculates the angles for the wheels is [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/205755685f77300a27d16c9739b1e397e30811d9/gazebo_plugins/src/gazebo_ros_ackermann_drive.cpp#L518). Which seems correct (I've rechecked the math, it checks out). However, in both my tests driving by hand as in the math for the angles of the wheels the angular component of the cmd_vel message is treated as an angle.

This is incorrect. The message is a geometry_msg twist. It should express the angular **velocity** around the z axis. Not the angle itself. The angular velocity and the angle are not the same. This should be changed, right? Or at the very least there should be a new parameter that lets you choose whether to treat it as the (correct) angular **velocity** or the (incorrect) angle (for compatibility reasons).

Any thoughts on this? Am I crazy or is this plugin simply flawed?

PS I think there is also a mistake in the calculation for the angles of the wheels, where the left wheel is always the inside wheel and the right the outside wheel. This does not change depending on whether you turn left or right. However, I am not 100% sure on this issue.

SivertHavso, 2022-01-12:
I agree it's confusing and not technically correct, but I think it's better to either keep the geometry_msgs/msg/Twist with steering angle instead of angular velocity, or move to ackermann_msgs/msg/AckermannDrive temporarily until any new /cmd_vel conventions are introduced that better support ackermann steering (https://discourse.ros.org/t/is-twist-still-a-good-velocity-command-interface/13218 might be relevant). Moving to ackermann_msgs/msg/AckermannDrive would mean (preferably) adding support for the extra message fields for desired acceleration and jerk. It's also a non-standard message type. 

In my experience most car-like robots tend to use steering angle instead of steering angular velocity, either in the local planner/controller, or they'll convert the angular velocity from the Nav2 controller to a steering angle before sending to the low-level hardware controllers or simulator (Gazebo in this case). Stage, another simulator often used with ROS, misuses /cmd_vel for steering angles as well.

What Nav2 controller are you using? [Regulated pure pursuit](https://github.com/ros-planning/navigation2/tree/main/nav2_regulated_pure_pursuit_controller) only supports outputting angular velocity, but if it's [teb_local_planner](https://github.com/rst-tu-dortmund/teb_local_planner/tree/ros2-master/), it has a parameter `/cmd_angle_instead_rotvel` you can set to true if you want a quick fix.

SivertHavso, 2022-01-12:
I agree it's a bad design, but I think the problem is not just limited to this plugin. I would be wary of changing the default behaviour. Adding an option (parameter) as you suggest might be best.

Re: PS
Now when I looked at it, it did seem weird, but working out the values outputted from the algorithm it seems to work as it is supposed to.

Calculating the result of the formulas for a steering angle of 0.5 rad and arbitrary dimensions of 2.0 for wheel seperation and 3.0 for wheel base:

```
tanSteer = tan(0.5);
         = 0.546

left = atan2(tanSteer, 1.0 - wheel_separation_ / 2.0 / wheel_base_ * tanSteer);
left = atan2(0.546, 1.0 - 2.0 / 2.0 / 3.0 * 0.546)
left = 0.588

right = atan2(tanSteer, 1.0 + wheel_separation_ / 2.0 / wheel_base_ * tanSteer);
right = atan2(0.546, 1.0 + 2.0 / 2.0 / 3.0 * 0.546)
right = 0.433
```

This makes sense since we are turning 0.5 to the left, the left wheel will be the inner wheel and has to turn more. On the other hand if we try steering by -0.5 rad:

```
tanSteer = tan(-0.5);
         = -0.546
         
left = atan2(-0.546, 1.0 - 2.0 / 2.0 / 3.0 * -0.546)
left = -0.432

right = atan2(-0.546, 1.0 + 2.0 / 2.0 / 3.0 * -0.546)
right = -0.588
```

Which also makes sense because in this case the right wheel is the inner one and is turning more.

However, you might have discovered it flips the steering angle when you reverse, which you might not want (https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1132#issuecomment-711513715)