ros-simulation/gazebo_ros_pkgs #1358: [ROS 2] gazebo_ros_camera: data publish only happens after the subscriber is already created in dataspace, but not after

TSC21, 2022-02-22:
Hi there,

Me and my team are seeing a weird problem (with both Foxy and Galactic) where, while using Fast-DDS as the rmw, the camera data publishers only work when the subscribers are already created. I.e., e.g. if I have `ros2 topic echo /camera/depth/image_raw` already launched in the console or if I have RVIZ2 already subscribing to those topics. If in the other hand I launch first the sim with the depth camera and only start the subscriber nodes after, they will never receive any data.

This leads to me thinking there might be some problem on the way the ROS2 publishers are initialized in the [`gazebo_ros_camera` plugin](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_camera.cpp). Here's also the SDF defintion for reference:
```xml
    <link name="link">
      <inertial>
        <pose>0.01 0.025 0.025 0 0 0</pose>
        <mass>0.01</mass>
        <inertia>
          <ixx>4.15e-6</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>2.407e-6</iyy>
          <iyz>0</iyz>
          <izz>2.407e-6</izz>
        </inertia>
      </inertial>
      <sensor name="depth_camera" type="depth">
        <always_on>0</always_on>
        <update_rate>30</update_rate>
        <camera name="camera">
          <horizontal_fov>1.5009831567</horizontal_fov>
          <image>
            <format>R8G8B8</format>
            <width>640</width>
            <height>480</height>
          </image>
          <clip>
            <near>0.4</near> <!-- Intel RealSense D455 -->
            <far>20</far> <!-- Intel RealSense D455 -->
          </clip>
          <distortion>
            <k1>0.0</k1>
            <k2>0.0</k2>
            <k3>0.0</k3>
            <p1>0.0</p1>
            <p2>0.0</p2>
            <center>0.5 0.5</center>
          </distortion>
        </camera>
        <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
          <camera_name>camera</camera_name>
          <frame_name>camera_link</frame_name>
          <hack_baseline>0.07</hack_baseline>
          <min_depth>0.4</min_depth> <!-- Intel RealSense D455 -->
          <max_depth>20</max_depth> <!-- Intel RealSense D455 -->
        </plugin>
      </sensor>
    </link>
```

Now curiosity: **if you actually use CycloneDDS, the problem disappears (!)**, but of course this cannot be a problem for one of the main Tier 1 rmw's (and still the main one on Foxy, which is FastDDS). This was not a problem 4/5 months ago, so something had to be changed meanwhile so this is not working anymore.

I am also tagging here the FastDDS rmw maintainers/main contributors for visibility: @hidmic @audrow @MiguelCompany @clalancette.

Thanks in advance for having a look at this!

TSC21, 2022-02-25:
> @TSC21 Standard question... Does it behave differently from sources / from binaries?
> 
> We'll try to reproduce and get back to you

I have not tested this built from source.

TSC21, 2022-02-28:
@MiguelCompany any updates here?

TSC21, 2022-05-02:
@MiguelCompany friendly ping

bastianhjaeger, 2022-05-03:
Ping as well. Me and a colleague (@DanMesh) see this issue as well on Ubuntu 20.04 machines. Only the switch to CyclonDDS resolved it so far.

elanius, 2022-12-01:
I am struggling with this issue too. I am using `gazebo_ros_pkgs` with `PX-Autopilot`.
ROS_DISTRO: foxy
Ubuntu 18.04
Same behaviour with binaries and sources.

The problem should be somewhere in `gazebo_ros_factory.cpp` because when it was not used, topics were published normally. In PX4-Autopilot -> sitl_run.sh, I disabled the loading of this plugin, and it worked.
```diff
# To use gazebo_ros ROS2 plugins
if [[ -n "$ROS_VERSION" ]] && [ "$ROS_VERSION" == "2" ]; then
	# ros_args="-s libgazebo_ros_init.so -s libgazebo_ros_factory.so"
-	ros_args="-s libgazebo_ros_init.so -s libgazebo_ros_factory.so"
+       ros_args="-s libgazebo_ros_init.so"
else
	ros_args=""
fi
```