ros-simulation/gazebo_ros_pkgs #1359: gazebo_ros_pkgs in ROS 2 Rolling has not been released on Ubuntu Jammy

nuclearsandwich, 2022-02-23:
The ROS 2 Rolling Ridley distribution recently [migrated to Ubuntu 22.04 Jammy](https://github.com/ros/rosdistro/pull/32036) using the rosdistro migration tool.

This repository is one which failed to bloom during the migration because the `libgazebo11-dev` dependency is not yet available in Ubuntu Jammy. Once the Jammy repositories and the rosdep database are updated it should be possible to release this repository in Rolling again by running bloom normally.

Packages which are not released in Rolling may not be automatically added to future stable ROS 2 distributions.

chapulina, 2022-03-18:
Just a note that Gazebo classic won't be officially supported on Jammy. ROS users can use the Gazebo 11 version which will be available upstream from Ubuntu, and won't get updates. See the conversation here:

https://community.gazebosim.org/t/gazebo-classic-on-jammy/1248

There are 2 rosdistro PRs adding Gazebo rosdep keys for Jammy:

* https://github.com/ros/rosdistro/pull/31560
* https://github.com/ros/rosdistro/pull/32481

chapulina, 2022-03-25:
The rosdep keys are ready. Now we're [blocked on the conflicts with OpenCV](https://build.ros2.org/job/Rpr__gazebo_ros_pkgs__ubuntu_jammy_amd64/5/console#console-section-10) (CC @j-rivero):

```
08:46:11 The following packages have unmet dependencies:
08:46:11  libopencv-core-dev : Depends: libtbb-dev but it is not installable
```

That should be fixed by [this patch](https://bugs.launchpad.net/ubuntu/+source/gazebo/+bug/1965780 upstream on Ubuntu.

Here's a list of the PRs open against the `ros2` branch which we may want to get in before blooming:

https://github.com/ros-simulation/gazebo_ros_pkgs/pulls?q=is%3Apr+is%3Aopen+base%3Aros2

chapulina, 2022-04-14:
Gazebo 11.10.2 version is now in the Jammy repository and the OpenCV conflict should be fixed. We should be ready to merge pending PRs and bloom.

chapulina, 2022-06-06:
This is done, `gazebo_ros_pkgs` is available for Humble and Rolling.

closed 2022-06-06 by chapulina