ros-simulation/gazebo_ros_pkgs #1361: [gazebo_ros] Compilation error in `test/test_gazebo_ros_init`

goekce, 2022-03-06:
During compilation of `gazebo_ros` in a chroot for building I get the following error. I do not get the error when I compile on my development environment.

Could this be related to a missing dependency like `ament_cmake_gtest`? I compile using `colcon`.

```
/usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o: in function `GazeboRosInitTest_Commands_Test::~GazeboRosInitTest_Commands_Test()':                                                                              test_gazebo_ros_init.cpp:(.text._ZN31GazeboRosInitTest_Commands_TestD2Ev[_ZN31GazeboRosInitTest_Commands_TestD5Ev]+0x10): undefined reference to `gazebo::ServerFixture::~ServerFixture()'                                                   /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o: in function `GazeboRosInitTest_Commands_Test::~GazeboRosInitTest_Commands_Test()':                                                                              test_gazebo_ros_init.cpp:(.text._ZN31GazeboRosInitTest_Commands_TestD0Ev[_ZN31GazeboRosInitTest_Commands_TestD5Ev]+0x14): undefined reference to `gazebo::ServerFixture::~ServerFixture()'                                                   /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o: in function `testing::internal::TestFactoryImpl<GazeboRosInitTest_Commands_Test>::CreateTest()':                                                                test_gazebo_ros_init.cpp:(.text._ZN7testing8internal15TestFactoryImplI31GazeboRosInitTest_Commands_TestE10CreateTestEv[_ZN7testing8internal15TestFactoryImplI31GazeboRosInitTest_Commands_TestE10CreateTestEv]+0x1e): undefined reference to `gazebo::ServerFixture::ServerFixture()'                                                                                                                                                                                                     /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTI17GazeboRosInitTest[_ZTI17GazeboRosInitTest]+0x10): undefined reference to `typeinfo for gazebo::ServerFixture'                                /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV17GazeboRosInitTest[_ZTV17GazeboRosInitTest]+0x28): undefined reference to `gazebo::ServerFixture::TearDown()'                                 /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV17GazeboRosInitTest[_ZTV17GazeboRosInitTest]+0x40): undefined reference to `gazebo::ServerFixture::Unload()'                                   /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV17GazeboRosInitTest[_ZTV17GazeboRosInitTest]+0x48): undefined reference to `gazebo::ServerFixture::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'                                                                                                                                                                                            /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV17GazeboRosInitTest[_ZTV17GazeboRosInitTest]+0x50): undefined reference to `gazebo::ServerFixture::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'                                                                                                                                                                                      /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV17GazeboRosInitTest[_ZTV17GazeboRosInitTest]+0x58): undefined reference to `gazebo::ServerFixture::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'                                                                                                                                   /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV17GazeboRosInitTest[_ZTV17GazeboRosInitTest]+0x60): undefined reference to `gazebo::ServerFixture::LoadArgs(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'                                                                                                                                                                                        /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV31GazeboRosInitTest_Commands_Test[_ZTV31GazeboRosInitTest_Commands_Test]+0x28): undefined reference to `gazebo::ServerFixture::TearDown()'     /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV31GazeboRosInitTest_Commands_Test[_ZTV31GazeboRosInitTest_Commands_Test]+0x40): undefined reference to `gazebo::ServerFixture::Unload()'       /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV31GazeboRosInitTest_Commands_Test[_ZTV31GazeboRosInitTest_Commands_Test]+0x48): undefined reference to `gazebo::ServerFixture::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'                                                                                                                                                                /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV31GazeboRosInitTest_Commands_Test[_ZTV31GazeboRosInitTest_Commands_Test]+0x50): undefined reference to `gazebo::ServerFixture::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'                                                                                                                                                          /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV31GazeboRosInitTest_Commands_Test[_ZTV31GazeboRosInitTest_Commands_Test]+0x58): undefined reference to `gazebo::ServerFixture::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)'                                                                                                       /usr/bin/ld: CMakeFiles/test_gazebo_ros_init.dir/test_gazebo_ros_init.cpp.o:(.data.rel.ro._ZTV31GazeboRosInitTest_Commands_Test[_ZTV31GazeboRosInitTest_Commands_Test]+0x60): undefined reference to `gazebo::ServerFixture::LoadArgs(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'                                                                                                                                                            collect2: error: ld returned 1 exit status                                                                                                                                                                                                   make[2]: *** [test/CMakeFiles/test_gazebo_ros_init.dir/build.make:210: test/test_gazebo_ros_init] Error 1                                                                                                                                    make[1]: *** [CMakeFiles/Makefile2:650: test/CMakeFiles/test_gazebo_ros_init.dir/all] Error 2
```

FallenDeity, 2023-10-08:
Did you find a solution for this?

goekce, 2023-10-08:
Not really. Currently I use Ubuntu Docker image with precompiled packages on my non-Ubuntu host.

goekce, 2023-10-08:
Given that this issue did not catch significant attention, I would like to close this issue.

closed 2023-10-08 by goekce