ros-simulation/gazebo_ros_pkgs #1366: wrong imu linear_acceleration.z with libgazebo_ros_imu_sensor.so plugin.

panecho, 2022-03-31:
**Issue:**
plot result showed as following picture, left is in gazebo, and right is ros topic in ros rqt.<pub topic at 200HZ>
[image]

**my imu sdf file:**
```
<?xml version="1.0" ?>
<sdf version="1.6">
  <model name="imu_ros">
    <link name="link">
      <gravity>1</gravity>
      <inertial>
        <mass>0.1</mass>
      </inertial>
      <visual name="visual">
        <geometry>
          <box>
            <size>0.1 0.1 0.1</size>
          </box>
        </geometry>
      </visual>
      <collision name="collision">
        <geometry>
          <box>
            <size>0.1 0.1 0.1</size>
          </box>
        </geometry>
      </collision>
      <sensor name="imu" type="imu">
        <always_on>1</always_on>
        <update_rate> 200 </update_rate>
        <visualize>true</visualize>
        <topic>imu_raw</topic>
        <plugin name="imu_plugin" filename="libgazebo_ros_imu_sensor.so">
          <topicName>/imu_raw</topicName>
          <frameName>imu_link</frameName>
          <updateRateHZ>200</updateRateHZ>
          <gaussianNoise>0.0</gaussianNoise>
          <xyzOffset>0 0 0</xyzOffset>
          <rpyOffset>0 0 0</rpyOffset>
        </plugin>
      </sensor>
    </link>
  </model>
</sdf>
```

KhalidOwlWalid, 2022-04-23:
By any chance, what are you currently working on?

https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1363

I am having the same issue. Do you mind trying this out and see if this also happens to you?

panecho, 2022-04-24:
> By any chance, what are you currently working on?
> 
> #1363
> 
> I am having the same issue. Do you mind trying this out and see if this also happens to you?

Actually, I didn't figure it out. But I got some info which may help you from my weekly debug.
1> Do not use gazebo_ros_pkg for imu simulation, imu simulation without ros api seem to more stable.
2> Try different physics engine install of ode, imu data seem more stable.
By the way, all engine got noise data for imu simulation in motion sittuation. Good luck.

KhalidOwlWalid, 2022-04-24:
By trying different physics engine, are you implying that I should try physics engine other than gazebo?

> 1> Do not use gazebo_ros_pkg for imu simulation, imu simulation without ros api seem to more stable.
I've read the source code of the IMU for and as far as I am concern, here is where they are [calculating the IMU sensor data.](https://github.com/osrf/gazebo/blob/2e42eb91e7638da54b36dbb8fdd4ae5e3a310d6d/gazebo/sensors/ImuSensor.cc#L348-L375).  By `imu simulation without ros api seem to be more stable` , are you implying that the plugin uses ROS API? Cause as far as I am concern, I dont think that they are using any ROSS API in the plugin.

panecho, 2022-04-25:
> 
when you lauch gazebo by 
`roslaunch gazebo_ros empty_world.launch` for ros support, it will load a system plugin which I named ros api, [see here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_ros/src/gazebo_ros_api_plugin.cpp) which affect the imu data athought i agree with the imu implementation you mentioned,, actually I also review this code before.

for different physics engine mean launch gazebo by `gaebo -e ode | bullet | simbody | dart`, add imu, you will find different imu performance with different physics engine.

robilnovice, 2023-01-17:
did you try changing real time update and max step size accordingly to 10000, 0.0001 ?

In my simulation if I change it to 500 hz I get wrong imu readings

EmmanuelMess, 2024-10-23:
Adding noise helps with this. ~0.5 gaussian noise.