ros-simulation/gazebo_ros_pkgs #1368: [BUG] calling `/reset_simulation` causes `/xx/joint_states` stop.

ZhenshengLee, 2022-04-02:
## ENV

- ubuntu 20.04
- ros noetic
- gazebo 11.9.0

## reproduce

roslaunch a gazebo configuration with joint_state_publisher and robot_state_publisher

```sh
rosservice call /gazebo/reset_simulation "{}"
```

## actual behavior

```sh
rostopic echo /xxx/joint_states
```

stop outputing.

## expected behavior

```sh
rostopic echo /xxx/joint_states
```

works and output msgs.

## other info

may be related with #1347 #169

DongYanqi1029, 2022-04-07:
same problem with my /imu sensor

ba1Sta, 2022-08-08:
met the same problem

fuhuaRobotic, 2023-02-14:
the same with my imu topic, I found it seems the imu output data cannot cover the virtual time it has passed, it can only restart the output after the reset timestamp.