ros-simulation/gazebo_ros_pkgs #1370: NO gazebo_ros_openni_kinect.cpp in /opt/ros/melodic

Gaviiin-W, 2022-04-20:
I'm new to ROS and I'm confused with the DEBUG publishing message when I use the kinect camera in ROS as following:

[image]

I DO want to stop the DEBUG message and I think what I need to do is deleting the **[ROS_DEBUG]("publishing default camera info, then openni kinect camera info");**

But I can't find the relatived gazebo_ros_openni_kinect.cpp  but only the libgazebo_ros_openni_kinect.so in /opt/ros/melodic/lib

How can I deal with this problem?or maybe some tutorials about it?

quarkytale, 2022-04-22:
You can use the [ROS rqt_logger_level](http://wiki.ros.org/rqt_logger_level) tool to disable all debug messages.
Also, the `gazebo_ros_openni_kinect.cpp` can't find the file probably because it's a binary installation. To view source files, [install from source](http://wiki.ros.org/noetic/Installation/Source) or checkout on [github](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_plugins/src/gazebo_ros_openni_kinect.cpp)!

closed 2022-04-22 by quarkytale

Gaviiin-W, 2022-04-22:
Thank u ,**quarkytale**!
I used the rqt_logger_level before and that quite worked.But I need to set the rqt_logger_level every time I roslaunch it.Is it possible to configure it?

Gaviiin-W, 2022-04-22:
> You can use the [ROS rqt_logger_level](http://wiki.ros.org/rqt_logger_level) tool to disable all debug messages. Also, the `gazebo_ros_openni_kinect.cpp` can't find the file probably because it's a [...]

Thank u ,**quarkytale**!
I used the rqt_logger_level before and that quite worked.But I need to set the rqt_logger_level every time I roslaunch it.Is it possible to configure it?

quarkytale, 2022-04-22:
This should help configuring what you need: [rosconsole](http://wiki.ros.org/rosconsole#Configuration). I'll suggest checking out [ROS Answers](https://answers.ros.org/questions/) too. Hope it helps!