ros-simulation/gazebo_ros_pkgs #1372: Combination Ubuntu Jammy / ROS Rolling have undefined tbb symbols

j-rivero, 2022-04-25:
On Ubuntu Jammy + ROS Rolling, the pr job  [is failing due to problems resolving tbb symbols](https://build.ros2.org/job/Rpr__gazebo_ros_pkgs__ubuntu_jammy_amd64/13):

```
14:03:12 [ 58%] Linking CXX executable test_gazebo_ros_factory
14:03:12 /usr/bin/ld: CMakeFiles/test_gazebo_ros_factory.dir/test_gazebo_ros_factory.cpp.o: undefined reference to symbol '_ZN3tbb6detail2r114notify_waitersEm'
14:03:12 /usr/bin/ld: /lib/x86_64-linux-gnu/libtbb.so.12: error adding symbols: DSO missing from command line
14:03:12 collect2: error: ld returned 1 exit status
14:03:12 gmake[2]: *** [test/CMakeFiles/test_gazebo_ros_factory.dir/build.make:254: test/test_gazebo_ros_factory] Error 1
14:03:12 gmake[1]: *** [CMakeFiles/Makefile2:621: test/CMakeFiles/test_gazebo_ros_factory.dir/all] Error 2
14:03:12 gmake: *** [Makefile:146: all] Error 2
undefined reference to symbol '_ZN3tbb6detail2r114notify_waitersEm'
```

As the compiler says, the symbol is in `/lib/x86_64-linux-gnu/libtbb.so.12` so we need to declare it as a dependency in Gazebo11 CMake module so linking against the code is done without missing parts.

/cc @ahcorde

traversaro, 2022-04-25:
Related to:
* https://github.com/osrf/gazebo/issues/3008
* https://github.com/conda-forge/gazebo-feedstock/issues/121

I thought (erroneously, it seems) that it was exclusively a tbb >= 2021 problem so in https://github.com/osrf/gazebo/pull/3174 I fixed it just when tbb >= 2021 was used.

chapulina, 2022-05-02:
* Cross-reference to https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1359

j-rivero, 2022-05-03:
This is now fixed. Let's get a new release out.

commit 9954bb9f (2022-05-03): Include TBB in gazebo-dev cmake to fix #1372 (#1373)
closed 2022-05-03 by j-rivero