ros-simulation/gazebo_ros_pkgs #1374: CMake 3.22 build trigger a warning about find_package_handle_standard_args

j-rivero, 2022-04-27:
While building the software in Jammy, I'm seeing a new warning:
```bash
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  /home/jrivero/code/ros2/ws_gazebo_ros_pkgs_jammy/install/gazebo_dev/share/gazebo_dev/cmake/gazebo_dev-extras.cmake:2 (find_package)
  /home/jrivero/code/ros2/ws_gazebo_ros_pkgs_jammy/install/gazebo_dev/share/gazebo_dev/cmake/gazebo_devConfig.cmake:41 (include)
  CMakeLists.txt:26 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.           
```

I think it is related to https://github.com/osrf/gazebo/blob/gazebo11/cmake/gazebo-config.cmake.in#L38 (line 72 in `/usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake`).

mikepurvis, 2023-02-24:
We're seeing this on NixOS 22.11 as well, with CMake 3.24.3.

traversaro, 2023-02-25:
I think it should be sufficient to change `include(FindPkgConfig)` to `find_package(PkgConfig)` .

traversaro, 2023-02-26:
> I think it should be sufficient to change `include(FindPkgConfig)` to `find_package(PkgConfig)` .

See PR https://github.com/gazebosim/gazebo-classic/pull/3301 .