ros-simulation/gazebo_ros_pkgs #1392: gazebo_ros2_control namespace

muttistefano, 2022-05-18:
Hi everyone, i am using ROS2 galactic with the debian repository packages.
I want to load multiple robots in multiple namespaces , each robot with its gazebo_ros2_control node.
Currently I'm am running the robot_state_publisher and gazebo spawner for each roboti in this way , changing the namespace for each robot accordingly:
```
robot_state_publisher_node_2 = Node(
        package="robot_state_publisher",
        executable="robot_state_publisher",
        namespace="sweepee_1",
        output="log",
        parameters=[robot_description_2,frame_prefix_param_2],
    )

  spawn_sweepee_1 = Node(package='gazebo_ros', executable='spawn_entity.py',namespace="sweepee_1",
                          arguments=['-entity', 'sw1', "-topic", "robot_description",
                                     "-robot_namespace","sweepee_1",
                                     "-x"," 1"],
                          output='screen')
```
while the gazebo_ros2_control plugin is in the xacro this way:
```
    <gazebo reference="world">
    </gazebo>
    <gazebo>
      <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
        <robot_param>robot_description</robot_param>
        <!-- <parameters>/home/kolmogorov/Documents/2ROS/lampo_ws_ros2/install/lampo_description/share/lampo_description/config/ur_controllers_1.yaml</parameters> -->
        <parameters>$(arg simulation_controllers)</parameters>
        <robot_param_node>robot_state_publisher</robot_param_node>
      </plugin>
    </gazebo>
```
When I run the code, it seems like the namespace is neglected, and everything is loaded under the namespace of the first node, for instance, when i run it for 2 robots it gets:

```
/sweepee_1/front_laser_plugin
/sweepee_1/front_laser_plugin
/sweepee_1/gazebo_ros2_control
/sweepee_1/gazebo_ros2_control
```

Further plugin are also loaded under the wrong namespace.
Am i doing something wrong?
if someone points me a known problem i can actually collaborate.
Thanks

muttistefano, 2022-05-18:
I might have posted this in the wrong repo, sorry

tonynajjar, 2022-05-24:
@muttistefano I don't think you posted in the wrong repo.
I'm getting somewhat of the same behavior, for all plugins in fact. I pass a namespace into the xacro as such:

```
      <plugin name="backward_lidar_plugin" filename="libgazebo_ros_ray_sensor.so">
        <ros>
          <namespace>/$(arg namespace)</namespace>
          <argument>~/out:=backward_lidar</argument>
        </ros>
        <output_type>sensor_msgs/Range</output_type>
        <radiation_type>infrared</radiation_type>
        <frame_name>backward_lidar</frame_name>
      </plugin>
```

and for gazebo_ros2_control:

```
  <gazebo>
    <plugin name="gazebo_ros2_control_plugin" filename="libgazebo_ros2_control.so">
      <ros>
        <namespace>/$(arg namespace)</namespace>
        <remapping>~/cmd_vel:=cmd_vel_smoothed</remapping>
        <remapping>~/odom:=odom</remapping>
      </ros>
      <robot_param_node>robot_state_publisher</robot_param_node>
      <robot_param>robot_description</robot_param>
      <parameters>$(find amr_bringup)/config/control/$(arg namespace)_simulation_tricycle_controller.yaml</parameters>
    </plugin>
  </gazebo>
```

When running one instance, I get the correct namespace, but when running two instances, the plugins get the same namespace:

```
$ ros2 node list
/amr2/backward_lidar_plugin
/amr2/backward_lidar_plugin
/amr2/gazebo_ros2_control_plugin
/amr2/gazebo_ros2_control_plugin
```

I did some debugging and I could narrow down the issue to this [line](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/645e4d6565ea979eb408d17c8046475b8aa77069/gazebo_ros/include/gazebo_ros/node.hpp#L212): The namespace passed to std::make_shared<Node>() is different than the resulting namespace of the created node (node->get_namespace()). In fact it uses the namespace of the previously created node.

Would really appreciate some help here. Thanks

bobbleballs, 2023-02-24:
Hi I've looked into this at https://github.com/ros-controls/gazebo_ros2_control/issues/127
I think it's unrelated to gazebo_ros_pkgs, though they're definitely doing something that causes things to mess up!