ros-simulation/gazebo_ros_pkgs #1398: Same namespace for multiple plugins

tonynajjar, 2022-05-24:
When spawning multiple instances, the same namespace is used for all plugins, although a different namespace is passed each time

I pass a namespace into the xacro as such:

```
      <plugin name="backward_lidar_plugin" filename="libgazebo_ros_ray_sensor.so">
        <ros>
          <namespace>/$(arg namespace)</namespace>
          <argument>~/out:=backward_lidar</argument>
        </ros>
        <output_type>sensor_msgs/Range</output_type>
        <radiation_type>infrared</radiation_type>
        <frame_name>backward_lidar</frame_name>
      </plugin>
```

and for gazebo_ros2_control:

```
  <gazebo>
    <plugin name="gazebo_ros2_control_plugin" filename="libgazebo_ros2_control.so">
      <ros>
        <namespace>/$(arg namespace)</namespace>
        <remapping>~/cmd_vel:=cmd_vel_smoothed</remapping>
        <remapping>~/odom:=odom</remapping>
      </ros>
      <robot_param_node>robot_state_publisher</robot_param_node>
      <robot_param>robot_description</robot_param>
      <parameters>$(find amr_bringup)/config/control/$(arg namespace)_simulation_tricycle_controller.yaml</parameters>
    </plugin>
  </gazebo>
```

When running one instance, I get the correct namespace, but when running two instances, the plugins get the same namespace:

```
$ ros2 node list
/amr2/backward_lidar_plugin
/amr2/backward_lidar_plugin
/amr2/gazebo_ros2_control_plugin
/amr2/gazebo_ros2_control_plugin
```

I did some debugging and I could narrow down the issue to this [line](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/645e4d6565ea979eb408d17c8046475b8aa77069/gazebo_ros/include/gazebo_ros/node.hpp#L212): The namespace passed to std::make_shared<Node>() is different than the resulting namespace of the created node (node->get_namespace()). In fact it uses the namespace of the previously created node.

Would really appreciate some help here. Thanks

muttistefano, 2022-05-24:
i think https://github.com/beta-robots has forked some repos such as ros2_control and ros2_controllers and partially fixed this part.
I found a comment in an issue that I'm not able to retrieve now , with a list of the important commits.
I am still experiencing some problems but at least I'm able to load gazebo plugins in different namespaces.
I don't know if this fix is included in the rolling release.

tonynajjar, 2022-05-24:
I don't rhink the issue is in ros2_control though. It also happens with other plugins

bennoben, 2023-02-24:
#1392
Hi I've had a look into this and it's an issue with the gazebo_ros_control plugin trying to set namespaces of nodes that already have namespaces set.
I've gone into more detail in this comment at ros-controls/gazebo_ros2_control/issues/127
Would be good to see if this helps you as it fixed it for me.

bobbleballs, 2023-02-24:
Hi I've looked into this at ros-controls/gazebo_ros2_control/issues/127
Hopefully this fixes things for you also @tonynajjar