ros-simulation/gazebo_ros_pkgs #1406: compile "gazebo_ros” packages fail,a lot of probuf error，please help me

panrucheng5013, 2022-07-05:
at ubuntu 20.4 ,ros2 foxy  sources version,gazebo11,  I clone the “gazebo_ros_pkgs" source project to build，all packages compiled sucessfully except "gazebo_ros" packages，it displays the following error message:
error: ‘google::protobuf::uint8* gazebo::msgs::PoseAnimation::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  145 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/google/protobuf/message_lite.h:49,
                 from /usr/include/google/protobuf/generated_enum_util.h:36,
                 from /usr/include/google/protobuf/map.h:49,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:25,
                 from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:41,
                 from /home/panrucheng/gazebo_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_properties.cpp:15:

/usr/include/gazebo-11/gazebo/msgs/link.pb.h: In member function ‘google::protobuf::UnknownFieldSet* gazebo::msgs::Link::mutable_unknown_fields()’:
/usr/include/gazebo-11/gazebo/msgs/link.pb.h:113:55: error: no matching function for call to ‘google::protobuf::internal::InternalMetadata::mutable_unknown_fields()’
  113 |     return _internal_metadata_.mutable_unknown_fields();
      |                                                       ^
In file included from /usr/include/google/protobuf/message_lite.h:49,
                 from /usr/include/google/protobuf/generated_enum_util.h:36,
                 from /usr/include/google/protobuf/map.h:49,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /usr/include/gazebo-11/gazebo/msgs/poses_stamped.pb.h:25,
                 from /usr/include/gazebo-11/gazebo/physics/PhysicsTypes.hh:26,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:41,
                 from /home/panrucheng/gazebo_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_properties.cpp:15:
....
     a large number of  probuf error messages, when i reinstall "libgazebo11-dev" and “”libprotobuf-dev" ,it still show this probuf error

talregev, 2022-08-22:
Happen for me also. I think this repo is not maintain anymore. 
This error occur due the fact that the protobuf is updated to 3.12.4, and this package that compile with protobuf, didn't update accordingly. 
https://lists.fedoraproject.org/archives/list/robotics@lists.fedoraproject.org/thread/3RM37X2WPUMJIW7Y2LZ4TNQ4A6I6EEZE/
https://www.spinics.net/linux/fedora/fedora-robotics/msg00492.html

panrucheng5013, 2022-09-21:
> Happen for me also. I think this repo is not maintain anymore. This error occur due the fact that the protobuf is updated to 3.12.4, and this package that compile with protobuf, didn't update [...]

Thanks a lot,I rebuild the system protobuf version 3.6.1  to solve it