ros-simulation/gazebo_ros_pkgs #1411: Launch parameter errors in gzserver.launch.py

scpeters, 2022-07-29:
I'm seeing some weird console errors using the `ros2` branch with the following launch command:

~~~
$ ros2 launch gazebo_ros gzserver.launch.py verbose:=true physics:=ode
...
[Err] [Server.cc:472] Could not open file[]
[Wrn] [Server.cc:381] Falling back on worlds/empty.world
...
[Err] [Server.cc:530] Unregistered physics engine [ ], the default will be used instead
~~~

jacobperron, 2022-07-29:
Reverting https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1376 seems to resolve the issue.
I haven't looked too closely at what we can do to avoid using `shell=True` while avoiding these noisy console errors at the same time.

jacobperron, 2022-10-04:
This was fixed in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1414

PR #1414 (2022-09-27): Fix console errors and warnings when launching gzserver
Fixes #1411 

Since we are no longer launching with `shell=True`, we do not need the extra spaces between command arguments (d369be816b0f7d21b44591d163871ba81a0dab90).
In particular, they were causing an error to be printed to the console if the `physics` option was given.

If one of the substitutions in the gzserver command results in an empty string,
then we see the following console error and warning:

    [Err] [Server.cc:472] Could not open file[]
    [Wrn] [Server.cc:381] Falling back on worlds/empty.world

This is because gzserver  [... cut]
closed 2022-10-04 by jacobperron