ros-simulation/gazebo_ros_pkgs #1412: /clock frequency can not be changed when using gzserver.launch.py

Arne-Christoph, 2022-07-30:
Hey all!

I believe this issue is related to 
#1411 and #1376 .

If I use the changes introduced by #1376 and try to change the publish frequency of the /clock topic via the extra_gazebo_args parameter of gzserver.launch.py
e.g.
"extra_gazebo_args": "--ros-args --params-file gazebo_params.yaml",

as e.g. proposed here: https://answers.ros.org/question/390701/change-clock-rate-in-ros-2-gazebo/
this is no longer working. 

If I change the shell parameter back to true it is working again. Does someone has a clue what is going wrong?

shonigmann, 2023-01-06:
Just ran into a similar issue trying to run the gazebo client with a SystemPlugin (we want to run the equivalent of `gazebo -g libgazebo_system_plugin.so` so we pass `-g libgazebo_system_plugin.so` to the `extra_gazebo_args` argument). 

I had the same experience that changing to `shell=True` returned things to working as expected, at the penalty of things sometimes not shutting down as expected. As an alternative, per https://github.com/ros2/launch/issues/263, if I split the string into a list (hardcoded for the sake of testing) things do work as expected with `shell=False`. 

Can't think of a clean way to parse a string `LaunchConfiguration` into a list and I don't think it is possible to pass a list of strings in to a Launch Argument directly... For now our workaround is have our own copy of the `gzclient.launch.py` `ExecuteProcess` logic in our own launch file to have full control over what we put in the `cmd` but it doesn't feel ideal...

meyerj, 2023-09-15:
Is there a recommended workaround for this issue, ideally without having to fork and build `gazebo_ros_pkgs` or copy the launch files in package `gazebo_ros`?

A clock publishing rate of 10 Hz is far too coarse to run timers or periodic rates at higher frequencies in external nodes with `use_sim_time`, and with the versions released into ROS Humble I cannot find a way to override the `publish_rate` parameter via the `extra_gazebo_args:=` roslaunch argument as explained [here](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-Migration:-ROS-Clocks-and-sim-time#clock-rate).

_Update:_ What still works is the approach with an extra configuration file and the `params_file` roslaunch argument.

Related pull requests and issues: #1211, #1214