ros-simulation/gazebo_ros_pkgs #1439: ROS Humble binary instalation for arm64

renzobazan96, 2022-11-22:
Hello,

I would like to install the pacakge over `sudo apt install ros-gazebo-ros-pkgs`. Currently there exist a package for amd64 but no for arm64: http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-humble-gazebo-ros-pkgs/
For now I can build it from source.

Are there any plans to release this package for humble-arm64?

Keshava2k01, 2024-05-08:
Is there any updates for this? I'm also trying to install gazebo-ros packages in vm in arm64 pc

j-rivero, 2024-05-10:
> Is there any updates for this? I'm also trying to install gazebo-ros packages in vm in arm64 pc

Please check the latest comment on https://github.com/gazebosim/gazebo-classic/issues/3236#issuecomment-1331053760 . In summary: there are no packages and probably at this point the chances of having them are really remote. Best option could be to use the https://launchpad.net/~openrobotics/+archive/ubuntu/gazebo11-non-amd64 to get gazebo11 on ARM and build ros_gazebo_pkgs from source.

closed 2024-05-10 by j-rivero