ros-simulation/gazebo_ros_pkgs #1442: Spawn_entity waiting for service /spawn_entity error "SOLVED"

esamet, 2022-12-14:
**Hi ı had a this error and ı find the solution to kill the gzserver, easy but annoying**

[INFO] [1671024686.152111830] [spawn_entity]: Waiting for service /spawn_entity
[ERROR] [1671024716.199868453] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[ERROR] [1671024716.201138555] [spawn_entity]: Spawn service failed. Exiting.

yudonghou123, 2022-12-22:
Hi,I have the same error, this error also bothers me, can you please solve it?

yudonghou123, 2022-12-23:
> **Hi ı had a this error and ı find the solution to kill the gzserver, easy but annoying**
>
> [INFO] [1671024686.152111830] [spawn_entity]: Waiting for service /spawn_entity [ERROR] [...]

Hi,I have the same error, this error also bothers me, can you please solve it?

zeoneo, 2023-11-19:
I am getting the same error, can someone help with debug hints ?

ROS_VERSION = humble
OS: Ubuntu 22.04
Python: 3.10

```
ros2 launch turtlebot3_gazebo spawn_turtlebot3.launch.py
[INFO] [launch]: All log files can be found below /home/$user/.ros/log/2023-11-19-09-29-34-235122-zeo-15594
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [spawn_entity.py-1]: process started with pid [15595]
[spawn_entity.py-1] [INFO] [1700366374.560103776] [spawn_entity]: Spawn Entity started
[spawn_entity.py-1] [INFO] [1700366374.560348433] [spawn_entity]: Loading entity XML from file /opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-1] [INFO] [1700366374.567405691] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-1] [INFO] [1700366374.567689008] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-1] [ERROR] [1700366404.643885794] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-1] [ERROR] [1700366404.644390847] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-1]: process has died [pid 15595, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity burger -file /opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x 0.0 -y 0.0 -z 0.01 --ros-args'].

```

JohnJackal, 2024-07-26:
I encountered this problem too. I'm using Ubuntu 22.04 on WSL 2, and my ROS version is humble. Python: 3.10.12.
The command I executed:
`export ROS_DOMAIN_ID=1
export TURTLEBOT3_MODEL=burger
export GAZEBO_MODEL_PATH=`ros2 pkg \
prefix turtlebot3_gazebo`/share/turtlebot3_gazebo/models/
gazebo -s libgazebo_ros_init.so -s libgazebo_ros_factory.so myworld.world
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py`
The following is the execution result.
`[INFO] [launch]: All log files can be found below /home/johnjackal/.ros/log/2024-07-26-15-53-40-951403-L-Zhenke-Zhang-137166
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [137196]
[INFO] [gzclient-2]: process started with pid [137198]
[INFO] [robot_state_publisher-3]: process started with pid [137200]
[INFO] [spawn_entity.py-4]: process started with pid [137202]
[robot_state_publisher-3] [INFO] [1721980421.750842730] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1721980421.750951729] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1721980421.750959121] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1721980421.750961834] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1721980421.750964522] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1721980421.750966859] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1721980421.750969372] [robot_state_publisher]: got segment wheel_right_link
[ERROR] [gzserver-1]: process has died [pid 137196, exit code 255, cmd 'gzserver /opt/ros/humble/share/turtlebot3_gazebo/worlds/turtlebot3_world.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[spawn_entity.py-4] [INFO] [1721980422.465467367] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1721980422.465996156] [spawn_entity]: Loading entity XML from file /opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1721980422.466921740] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1721980422.467369513] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [ERROR] [1721980452.520237767] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-4] [ERROR] [1721980452.520538188] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-4]: process has died [pid 137202, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity burger -file /opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x -2.0 -y -0.5 -z 0.01 --ros-args'].`