ros-simulation/gazebo_ros_pkgs #1465: 'colcon build failed' gazebo_ros

onurulusoy4, 2023-03-13:
## Issue Description

I am trying to install gazebo_ros_pkgs on ROS2 `ros2` branch but encountering issues during the `colcon build` process. The error message I receive is as follows:
```
Failed <<< gazebo_ros [48.0s, exited with code 2]

Summary: 6 packages finished [1min 37s]
1 package failed: gazebo_ros
4 packages had stderr output: cv_bridge gazebo_msgs gazebo_ros opencv_tests
2 packages not processed
```

It is always stuck at 81% building gazebo_ros, and also throws some warning installing the other packages.
Full error message: 
[gazebo_ros_error.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/10962592/gazebo_ros_error.txt)

[image]

## System Information
- OS: Ubuntu 22.04
- ROS2 Distribution: Humble
- gazebo_ros_pkgs Branch: `ros2` (Since there is no humble branch)
- vision_opencv Branch: `humble`

## Steps to Reproduce
1. Clone the gazebo_ros_pkgs repository on the `ros2` branch
2. Install all dependencies
3. Run `colcon build`

## Expected Behavior
The `colcon build` command should successfully build all packages, including gazebo_ros.

## Actual Behavior
The `colcon build` process fails with an error in the gazebo_ros package.

## Additional Information
I have tried the following:
- Checked all dependencies required by gazebo_ros_pkgs
- Tried building the `gazebo_ros_pkgs` package on the different branches.
- Disabled conda environment.

However, the issue still persists. Any help to resolve this issue would be appreciated. Thank you!

doza210, 2023-03-22:
it could be because the cmake files are using catkin whereas you are using ros2, try replacing with [ament](https://design.ros2.org/articles/ament.html) instead

onurulusoy4, 2023-03-26:
I found out that I got no errors compiling C++ packages which ament is used in cmakelists, instead I got in python where there is no cmake files. What can I do to to resolve?
My python package has a default setup configuration:
```
setup(
    name=package_name,
    version='0.0.0',
    packages=[package_name],
    data_files=[
        ('share/ament_index/resource_index/packages',
            ['resource/' + package_name]),
        ('share/' + package_name, ['package.xml']),
    ],
    install_requires=['setuptools'],
    zip_safe=True,
    maintainer='onur',
    maintainer_email='xx',  
    description='TODO: Package description',
    license='TODO: License declaration',
    tests_require=['pytest'],
    entry_points={
        'console_scripts': [
            'xx = xx:main',
            'xx = xx:main',
        ],
    },

)
```
Thanks for your response! @doza210