ros-simulation/gazebo_ros_pkgs #1483: cmd_vel and odom toipcs are not visible when using diff drive in ros2 foxy

SaumyaRaj188, 2023-05-19:
File is included without an error, plugin shows up in gazebo but no cmd_vel nor odom can be seen
here is the code : 

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" >

    <gazebo>
        <plugin name="skid_steer_drive" filename="libgazebo_ros_diff_drive.so">
            
            <!-- Number of wheel pairs -->
            <num_wheel_pairs>2</num_wheel_pairs>
            
            <!-- wheels front -->
            <left_joint>lf_wheel_joint</left_joint>
            <right_joint>rf_wheel_joint</right_joint>

            <!-- wheels back-->
            <left_joint>lb_wheel_joint</left_joint>
            <right_joint>rb_wheel_joint</right_joint>

            <!-- kinematics -->
            <wheel_separation>0.8</wheel_separation>
            <wheel_separation>0.8</wheel_separation>

            <wheel_diameter>0.3</wheel_diameter>
            <wheel_diameter>0.3</wheel_diameter>

            <!-- Limits -->
            <max_wheel_torque>20</max_wheel_torque>
            <max_wheel_acceleration>1.0</max_wheel_acceleration>

            
            <!-- Output -->
            <command_topic>cmd_vel</command_topic>
            <odometry_topic>odom</odometry_topic>
            <odometry_frame>odom</odometry_frame>
            <robot_base_frame>base_link</robot_base_frame>

            <publish_odom>true</publish_odom>
            <publish_odom_tf>true</publish_odom_tf>
            <publish_wheel_tf>true</publish_wheel_tf>

        </plugin>
    </gazebo>

</robot>
```

closed 2023-05-19 by SaumyaRaj188

AtharvaWani3, 2023-11-19:
How did you debug this?

SaumyaRaj188, 2023-11-20:
> How did you debug this?

ROS2 Gazebo Plugin Files were missing

Installed it using these command:
```
sudo apt install ros-foxy-gazebo-ros-pkgs
sudo apt install ros-foxy-ros2-control ros-foxy-ros2-controllers ros-foxy-gazebo-ros2-control
```