ros-simulation/gazebo_ros_pkgs #1519: AttributeError: module 'em' has no attribute 'RAW_OPT'

ccw1003, 2023-12-12:
when I catkin build this ros package, I met the error:
AttributeError: module 'em' has no attribute 'RAW_OPT';
Does anyone know how to deal with this problem?Or anyone met this error before?
I will be very grateful if anyone can help me.
[image]

standmit, 2023-12-14:
solution:
https://stackoverflow.com/a/77656642

kamronkhakimov, 2024-05-22:
I solved it by:
`pip install empy=3.3.4`

covelus, 2024-07-03:
I think I probably accidentally updated all packages, and broke it again, as new empy packages (4.1) miss that RAW_OPT attribute. 
I got the error "This environment is externally managed", and I fixed it through:
`pip3 uninstall empy --break-system-packages`
`pip3 install empy==3.3.4 --break-system-packages`

priya050898, 2024-07-25:
problem is not resolved using pip install empy=3.3.4 
problem still exist
[image]

Coisni1127, 2024-12-14:
[image]
problem still exist

proffalken, 2024-12-15:
I hit this issue earlier on today.

In your workspace, remove `build`, `install`, and `logs` directories (probably only needs `build`, but you may as well make it completely fresh!), then re-run `colcon build`.

This worked for me.

RobinSchmid7, 2025-02-14:
> I solved it by: `pip install empy=3.3.4`

pip install empy==3.3.4