ros-simulation/gazebo_ros_pkgs #1520: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name spcbot

Sien3330, 2024-01-03:
I am using ROS1 melodic with python3 on Ubuntu 18.04 (gazebo9)
I'm working on the spcbot project [.](https://github.com/Geekgineer/spcbot/tree/master) For this, I executed the launch file:

```
<?xml version="1.0"?>

<launch>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find spcbot_gazebo)/worlds/complete_boxs.world"/>
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="recording" value="false"/>
    <arg name="debug" value="false"/>
  </include>

  <!-- Convert an xacro and put on parameter server -->
  <param name="robot_description" command="$(find xacro)/xacro.py '$(find spcbot_description)/urdf/spcbot.urdf.xacro'" />

  <!-- Spawn a robot into Gazebo -->
  <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" output="screen"
        args="-param robot_description -urdf -model spcbot
        -x 2.5 -y 2.5 -z 0" />

  <!-- Robots state publisher -->
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" />

  <!-- Starting Joint state publisher node which will publish the joint values   -->
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
</launch>
```
However, when I run it, I encounter the following error:

```

Error [Param.cc:293] Invalid boolean value
Error [parser.cc:406] parse as old deprecated model file failed.
Error Code 7 Msg: Error reading element <visualize>
Error Code 7 Msg: Error reading element <sensor>
Error Code 7 Msg: Error reading element <link>
Error Code 7 Msg: Error reading element <model>
Error Code 7 Msg: Error reading element <sdf>
process[joint_state_publisher-5]: started with pid [14514]
[INFO] [1704268663.964560, 10.883000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name spcbot
[ERROR] [1704268663.966848, 10.885000]: Spawn service failed. Exiting.
[spawn_urdf-3] process has died [pid 14356, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -param robot_description -urdf -model spcbot -x 2.5 -y 2.5 -z 0 __name:=spawn_urdf __log:=/home/sien/.ros/log/7d12547a-aa07-11ee-be47-00155df7bf3a/spawn_urdf-3.log].
log file: /home/sien/.ros/log/7d12547a-aa07-11ee-be47-00155df7bf3a/spawn_urdf-3*.log
```

I tried fixing it by setting sim_time to 0 as suggested in this link [](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/864)but it didn't work. Any help would be appreciated.

Sien3330, 2024-01-03:
Gazebo is running, and the world is loading, but the model is not spawning! What could be the issue?

nov05, 2025-02-26:
```
Error [Param.cc:293] Invalid boolean value
Error [parser.cc:406] parse as old deprecated model file failed.
Error Code 7 Msg: Error reading element <visualize>
...
```

It is probably something with `spcbot.urdf.xacro`, the model definition.   
Can you ran the `check_urdf` command on that file and see what the output is?   

* https://answers.ros.org/question/386771/