ros-simulation/gazebo_ros_pkgs #1523: [ROS2]  [Err] [Master.cc:96] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.

Serarguy, 2024-03-01:
Hello, I'm new to ROS2 and I encounter an issue while following this [tutorial](https://www.youtube.com/watch?v=4QKsDf1c4hc&list=PLunhqkrRNRhYAffV8JDiFOatQXuU-NnxT&index=12&t=1026s) , after I launch my package I get this:

```
virtual-machine:~/dev_ws$ ros2 launch amr_platform launch_sim.launch.py verbose:=true

[INFO] [launch]: All log files can be found below /home/cosmin-pc/.ros/log/2024-03-01-22-27-01-879772-cosminpc-virtual-machine-30391
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [30394]
[INFO] [gzserver-2]: process started with pid [30396]
[INFO] [gzclient-3]: process started with pid [30398]
[INFO] [spawn_entity.py-4]: process started with pid [30404]
[robot_state_publisher-1] [INFO] [1709324822.559540708] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1709324822.560170002] [robot_state_publisher]: got segment caster_wheel_1
[robot_state_publisher-1] [INFO] [1709324822.560585211] [robot_state_publisher]: got segment caster_wheel_2
[robot_state_publisher-1] [INFO] [1709324822.560856379] [robot_state_publisher]: got segment caster_wheel_3
[robot_state_publisher-1] [INFO] [1709324822.565934270] [robot_state_publisher]: got segment caster_wheel_4
[robot_state_publisher-1] [INFO] [1709324822.566279702] [robot_state_publisher]: got segment chassis
[robot_state_publisher-1] [INFO] [1709324822.566621290] [robot_state_publisher]: got segment laser_frame
[robot_state_publisher-1] [INFO] [1709324822.566959204] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-1] [INFO] [1709324822.567169388] [robot_state_publisher]: got segment right_wheel
[gzclient-3] Gazebo multi-robot simulator, version 11.10.2
[gzclient-3] Copyright (C) 2012 Open Source Robotics Foundation.
[gzclient-3] Released under the Apache 2 License.
[gzclient-3] http://gazebosim.org
[gzclient-3] 
[gzserver-2] Gazebo multi-robot simulator, version 11.10.2
[gzserver-2] Copyright (C) 2012 Open Source Robotics Foundation.
[gzserver-2] Released under the Apache 2 License.
[gzserver-2] http://gazebosim.org
[gzserver-2] 
[spawn_entity.py-4] [INFO] [1709324823.622682035] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1709324823.623557866] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/iron/lib/python3.10/site-packages/rclpy/qos.py:313: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1709324823.634122478] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1709324823.647630200] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1709324823.649010971] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1709324823.670030870] [spawn_entity]: Calling service /spawn_entity
[gzserver-2] [Err] [Master.cc:96] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.
[gzserver-2] 
[gzserver-2] [Err] [Master.cc:96] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.
[gzserver-2] 
[ERROR] [gzserver-2]: process has died [pid 30396, exit code 255, cmd 'gzserver --verbose -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[gzserver-2] 
[gzserver-2] 
[gzclient-3] [Msg] Waiting for master.
[gzclient-3] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gzclient-3] [Msg] Publicized address: 192.168.137.129
[gzclient-3] [Msg] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[gzclient-3] [Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[gzclient-3] context mismatch in svga_surface_destroy
[gzclient-3] context mismatch in svga_surface_destroy

```

This is the launch file:

```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

from launch_ros.actions import Node

def generate_launch_description():

    # Include the robot_state_publisher launch file, provided by our own package. Force sim time to be enabled
    # !!! MAKE SURE YOU SET THE PACKAGE NAME CORRECTLY !!!

    package_name='amr_platform' #<--- CHANGE ME

    rsp = IncludeLaunchDescription(
                PythonLaunchDescriptionSource([os.path.join(
                    get_package_share_directory(package_name),'launch','rsp.launch.py'
                )]), launch_arguments={'use_sim_time': 'true'}.items()
    )

    # Include the Gazebo launch file, provided by the gazebo_ros package
    gazebo = IncludeLaunchDescription(
                PythonLaunchDescriptionSource([os.path.join(
                    get_package_share_directory('gazebo_ros'), 'launch', 'gazebo.launch.py')]),
             )

    # Run the spawner node from the gazebo_ros package. The entity name doesn't really matter if you only have a single robot.
    spawn_entity = Node(package='gazebo_ros', executable='spawn_entity.py',
                        arguments=['-topic', 'robot_description',
                                   '-entity', 'my_bot'],
                        output='screen')

    # Launch them all!
    return LaunchDescription([
        rsp,
        gazebo,
        spawn_entity,
    ])

```

It worked just fine until I "implemented" ros control. I tried creating a new VM with a fresh install of ROS2 and still get the same result. I also tried `killall gzserver` and still the same result. I'm a beginner so please keep that in mind while asking questions. Thanks!

closed 2024-03-06 by Serarguy

machineIntelligence, 2024-03-25:
Did you figure out this issue? I am having a similar problem.

Serarguy, 2024-03-26:
@machineIntelligence I fixed it but I did not find the exact fix or the reason why I encountered this issue. 
What I did was a reinstall of ros2, also a fresh VM, but the issue was still there. I did `killall gzserver gzclient` until I got the no server and client running, also added those to bash (cant find the exact link I followed to do this), restart the VM and it worked, didn't get the error since.

Danendra10, 2024-07-18:
Because you have solved it, i'll try to help explaining it, This is because there are still a gzsever running on background, this line shows it

```
Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.
```

If you find it again, you could use killall or pkill, or if you want to check it, you could use

```
ps aux | grep gz
```

Himanshukohale22, 2024-10-10:
i  encounter the same issue with my launch file but i could not able to resolve with restart or killall/pkill.

[image]

but my robot is spawn in gazebo and somehow its working as i want.