ros-simulation/gazebo_ros_pkgs #1529: tinyxml_vendor -> tinyxml2_vendor

tonynajjar, 2024-04-23:
I tried to build gazebo_ros on rolling and got:

```
Starting >>> gazebo_ros
--- stderr: gazebo_ros                                 
CMake Error at CMakeLists.txt:33 (find_package):
  By not providing "Findtinyxml_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "tinyxml_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "tinyxml_vendor"
  with any of the following names:

    tinyxml_vendorConfig.cmake
    tinyxml_vendor-config.cmake

  Add the installation prefix of "tinyxml_vendor" to CMAKE_PREFIX_PATH or set
  "tinyxml_vendor_DIR" to a directory containing one of the above files.  If
  "tinyxml_vendor" provides a separate development package or SDK, be sure it
  has been installed.
```

I guess it's related to https://discourse.ros.org/t/upcoming-api-break-in-urdfdom/34750? FYI @clalancette

azeey, 2024-04-23:
That's correct. `gazebo_ros` depends on Gazebo Classic, which will not be available on Ubuntu 24.04. See https://discourse.ros.org/t/gazebo-classic-end-of-life-ros-2-jazzy/36239

tonynajjar, 2024-04-23:
Okay makes sense, thank you

closed 2024-04-23 by tonynajjar