ros-simulation/gazebo_ros_pkgs #1536: libgazebo_ros_imu_sensor with ign gazebo

luxapana, 2024-07-13:
Hi all
I am trying to use libgazebo_ros_imu_sensor.so plugin with ros2. The relevent part of urdf is given below:
```
<gazebo reference="imu_link">
        <sensor name="imu_sensor" type="imu">
            <plugin filename="libgazebo_ros_imu_sensor.so" name="imu_plugin">
                <ros>
                    <namespace>/demo</namespace>
                    <remapping>~/out:=imu</remapping>
                </ros>
                <initial_orientation_as_reference>false</initial_orientation_as_reference>
            </plugin>
            <always_on>true</always_on>
            <update_rate>100</update_rate>
        </sensor>
    </gazebo>
```

However, when launched through ros_gz_sim, I do not see any topics such as /demo. I am using ign gazebo. I am getting messages saying libgazebo_ros_imu_sensor is not exposing any symbols.

what could I be doing wrong? Do these plugins work with ign gazebo?

Thanks

Muhammad540, 2024-10-01:
the namespace should be same as the name you provide to the plugin, in this case it should be "imu_plugin", if you want demo then change the plugin name to "demo"

octavioaguila, 2025-02-19:
Solution is [here](https://robotics.stackexchange.com/a/112322/33182)