ros-simulation/gazebo_ros_pkgs #1540: "apply_link_wrench" for a specific duration not working

Pratham-Pandey, 2024-07-23:
## Environment
* OS Version: Ubuntu 22.04 
* Build: Binary ROS2 Iron
* Gazebo Version: Gazebo Classic 11

## Description
* Expected behavior: I want to apply force to an object in gazebo. I am calling "/apply_link_wrench" service through terminal. When I am setting duration in the command to -1(means apply force continuously), things are working fine but when i set it to a non-negative number like 2(means apply force for  2 seconds), the object is not moving at all.

* Actual behavior: I expect the object to move for the amount of seconds specified in duration.

## Steps to reproduce

1. Launch gazebo through terminal using the following command:
```gazebo```

2. Next add a box using Gazebo GUI.

3. Make sure "/apply_link_wrench" service is available by running:
```ros2 service list```

4. Then run the following command:
``` ros2 service call /apply_link_wrench gazebo_msgs/srv/ApplyLinkWrench '{link_name: "unit_box::link", reference_frame: "", reference_point: { x: 100, y: 0, z: 0 }, wrench: { force: { x: 10, y: 0, z: 0 }, torque: { x: 0, y: 0, z: 0 } }, start_time: {sec: 0, nanosec: 0}, duration: {sec: 10, nanosec: 0} }' ```

## Output
When running the commands above, the object is not moving.

barbero77, 2024-11-17:
same problem...there is a solution? or is necessary to create a script to manage execution times

Pratham-Pandey, 2024-11-20:
@barbero77 As a workaround you could apply force to the object using "/apply_link_wrench" and then call "/clear_link_wrenches"  after desired amount of time to clear any force on the object.