ros-simulation/gazebo_ros_pkgs #1554: cant launch  gazebo_ros

reza0ramesh, 2024-11-02:
hi , i have problem using gazebo_ros package , when i want to launch a file with gazebo_ros pakage like : 
ros2 launch gazebo_ros gazebo.launch.py 
it doesnt work and there is not any error either :
[INFO] [launch]: All log files can be found below /home/reza720/.ros/log/2024-11-02-19-20-40-342948-reza720-GF63-Thin-10SC-5785
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [5786]
[INFO] [gzclient-2]: process started with pid [5788]
its all i see in terminal the point is i can launch an empty gazebo world with writing gazebo in terminal but i cant launch files and sometimes i even have error such as :
[INFO] [launch]: All log files can be found below /home/reza720/.ros/log/2024-11-02-19-13-40-486219-reza720-GF63-Thin-10SC-4585
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [4586]
[INFO] [gzclient-2]: process started with pid [4588]
[ERROR] [gzserver-1]: process has died [pid 4586, exit code 255, cmd 'gzserver -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[gzclient-2] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion px != 0' failed.
[ERROR] [gzclient-2]: process has died [pid 4588, exit code -6, cmd 'gzclient'].
is anybody have the same problem or any solution to my problem ?

osama13robot, 2024-11-14:
I have exactly the same problem as yours and I don't know what to do !!

mengxyokok, 2025-01-02:
ros2 launch gazebo_ros gazebo.launch.py 
[INFO] [launch]: All log files can be found below /home/mxy/.ros/log/2025-01-02-10-23-44-423563-kuanke-3202998
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [3203015]
[INFO] [gzclient-2]: process started with pid [3203017]

the same question